mriscoc/Ender3V2S1 #309: XYZ_position is blank after Homing and turning off LCD

classicrocker883, 2022-08-25:
**Describe the bug**

printer position at bottom of LCD are blank after getting values and turning off LCD then refreshing

**To Reproduce**
Steps to reproduce the behavior:
After Homing
1. Go to 'Main Menu' 
2. Click on 'Advanced Settings'
3. Scroll down to 'Turn off LCD'
4. Move or click encoder knob to reinitialize screen
5. XYZ coordinates are blank

**Expected behavior**
they should still remain as they were before LCD went off

**Version (please complete the following information):**
Firmware version and release date
### newest releast 427UBL

[image]

mriscoc, 2022-08-25:
Thanks for the report, it is currently a known issue fixed in the latest experimental currently under test, this bug is caused because the interface is waiting for a change in the axis values before updating the screen (drawing optimization). You will be the value redrawn if the axis moves.

classicrocker883, 2022-08-28:
I'm almost certain this issue has something to do with the dwin.cpp file.

I was moving some menu items around and for some reason I got an error say the #endif to `#if ENABLED(DWIN_LCD_PROUI)` has no #if.
it was able to compile when I removed the last #endif.

I tried looking through the file to see if there was another #endif for enabled (DWIN_LCD_PROUI)

closed 2022-08-28 by mriscoc

classicrocker883, 2022-08-28:
I will see if I can copy files over to reproduce or see if the problem goes away and pinpoint what it may be causing it.

mriscoc, 2022-08-28:
> it may have something to do with UBL being active. because this bug does not occur in the original Marlin firmware version of proui

No, this error was introduced recently due to drawing optimization but it was already fixed in the current experimental.

classicrocker883, 2022-08-29:
was it due to the encoder? i have shortened the list of probable causes and i think now it has to do with the encoder because other versions only allowed you to press the encoder (ENC) to get back from the LCD off state.

i like that with this firmware you can turn the knob or press the button.

im excited for the next release , do you have an ETA?

mriscoc, 2022-10-04:
Fixed in latest version.