mriscoc/Ender3V2S1 #316: NOZZLE_PARK_POINT doesnt work in config.h file

classicrocker883, 2022-08-26:
### HERE IS THE FIX

### (config.h line 2256)

```
#define NOZZLE_PARK_POINTS { { 0, 220, 40 } }
```

changed definition by adding an S and {}

### (proui.h line 44)

```
#define DEF_NOZZLE_PARK_POINT NOZZLE_PARK_POINTS
//{0, 220, 40}
```

removed the arbitrary value to the referenced one with added S

mriscoc, 2022-08-26:
What file is config.h? Maybe it is Configuration.h? What bug does this change fix? please elaborate.

classicrocker883, 2022-08-27:
yea configuration.h sorry about that it's less text to type

classicrocker883, 2022-08-27:
so if you wanted to change the park point values in the configuration file, it doesn't actually change them because it's not actually being called by `DEF_NOZZLE_PARK_POINT`
if you see those values are arbitrarily put in proui.h

so what this does is call for those values if u want to change them in the configuration file, otherwise it is pointless and does nothing

classicrocker883, 2022-08-27:
it may not need the S add and just be defined as `NOZZLE_PARK_POINT`
I havent tested that, Im don't fully know C++/C

mriscoc, 2022-08-27:
You need to do these changes:
```cpp
// In configuration.h:
#define NOZZLE_PARK_POINT { 0, 220, 40 } 

// In proui.h:
constexpr xyz_int_t DEF_NOZZLE_PARK_POINT = NOZZLE_PARK_POINT;

// In dwin.cpp
PRO_data.Park_point = DEF_NOZZLE_PARK_POINT;
```
I'm going to implement them in the next version.

classicrocker883, 2022-08-28:
@mriscoc hey BTW if i recall correctly, there is a typo in language_en.h

i think its MSG_NOOZLE....ect 
instead it should be NOZZLE

mriscoc, 2022-08-28:
It is MSG_NOOZLE_CLEAN but I'm not using that.

classicrocker883, 2022-08-28:
that fix worked great thanks for that. yea i dont think that msg was used

mriscoc, 2022-10-04:
Fixed in latest version.

closed 2022-10-04 by mriscoc