mriscoc/Ender3V2S1 #323: Display turn white after restart a print

Ice-Cue, 2022-08-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to print and cancel it again and start print again

**Expected behavior**
Display turns slowly white and is the button isnt working. have to restart the printer

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
newest version

mriscoc, 2022-08-27:
Sorry but we were unable to reproduce this issue.

mriscoc, 2022-08-31:
Check your connections, publish the printer, the exact firmware version used and the steps to reproduce the fault.

chrisgwynne, 2022-09-02:
Have this issue with; SYNWIT screen.

Had to go back to JYERS as you don't support this screen.

xdujiang, 2022-12-23:
Same issue here. The display gradually goes full white when enter the printing menu for the second time, and everything is not responding. Only a reboot can get it normal.
.
- Ways to reproduce:
1. enter the media menu
2. choose a gcode file
3. choose print or cancel
4. choose a gcode file again

- Software version
Firmware version: 20221222, 20221002. Both the same.
Screen firmware version: Packed in Ender-3 V2_HW_V4.2.2_SW_V1.0.7 2022-08-19

- Screen board

 
[image]

- Main chip
[image]

- Abnormal display
[image]
[image]

mriscoc, 2022-12-23:
Is your printer connected via USB?

xdujiang, 2022-12-23:
> Is your printer connected via USB?

nope, the only cable connected to the machine is power cable 😀

Ice-Cue, 2022-12-23:
I fixed it by just using octoprint and not printing over the display

mriscoc, 2022-12-24:
strange behaviour. So, printing from Octoprint works fine, but selecting a file doesn't? From the images, the screen starts to go white from the file selection menu and does not enter the printing screen.

mriscoc, 2022-12-24:
strange behavior. So, printing from Octoprint works fine, but selecting a file doesn't? From the images, the screen starts to go white from the file selection menu and does not enter the printing screen.

Ice-Cue, 2022-12-24:
It’s a 50/50 chance that it works to print from the display. The crazy’s thing is thats on 2 endet 3s from 2 different seller’s

xdujiang, 2022-12-24:
> I fixed it by just using octoprint and not printing over the display

I'm using the same temporary fix, but I want to keep the capability to print offline.

> strange behavior. So, printing from Octoprint works fine, but selecting a file doesn't? From the images, the screen starts to go white from the file selection menu and does not enter the printing screen.

Yes, everything OK if printing from octoprint. 
While printing from the display, printing screen can only be entered one time after the printer boots up. Whether printing or cancel, the secong time entering the printing screen cause a white screen.

> It’s a 50/50 chance that it works to print from the display. The crazy’s thing is thats on 2 endet 3s from 2 different seller’s

I'm running ender3 v2, and the issue can always be reproduced.

mriscoc, 2022-12-24:
OK, maybe the QR-code routine or preview routine corrupt the SYNWIT display, are you able to see the QR-code?

xdujiang, 2022-12-24:
> OK, maybe the QR-code routine or preview routine corrupt the SYNWIT display, are you able to see the QR-code?

The QR code with text "How to generate a previre" below?
Yes, it can be displayed at the first time.

mriscoc, 2022-12-24:
Please post a video, I will try to catch where the display is being affected.

xdujiang, 2022-12-24:
> Please post a video, I will try to catch where the display is being affected.

https://user-images.githubusercontent.com/10143140/209425291-f18bae5a-363c-468a-9300-5e511411195d.mp4

mriscoc, 2022-12-24:
When the screen goes to white, is the printer responding to Octoprint commands?

xdujiang, 2022-12-24:
> When the screen goes white, does the printer still respond to Octoprint commands? 

Yes, terminal is outputting consistently, and axis control is functional.

> This firmware has the preview and qr-code removed: [Ender3V2-422-BLTUBL-MPC-NOPREV-20221224.zip](https://github.com/mriscoc/Ender3V2S1/files/10298148/Ender3V2-422-BLTUBL-MPC-NOPREV-20221224.zip)

I flashed this firmware, manually tested “printing-stop-printing-stop” cycle for few times, it seems OK and no white screen during the test.

mriscoc, 2022-12-25:
This issue is specific to SYNWIT screens units, it is recommended that users with this display compile the firmware with the option `HAS_GCODE_PREVIEW` disabled in the Configuration.h file.

Ice-Cue, 2022-12-25:
Is it possible that you compile it olso for us?

2022 um 19:29:

mriscoc, 2022-12-25:
> […](#)

There is the possibility of request custom and sponsored compiles as is explained here: https://github.com/mriscoc/Special_Configurations#these-special-configurations-and-releases-are-sponsored-by-donors

Ice-Cue, 2022-12-25:
Ok, thx

tomjschwanke, 2022-12-25:
Thank you, I was running into the same issue on my Ender 3 V2 (also with a Synwit / Viewe display) and was able to fix it by disabling the preview stuff.

xdujiang, 2022-12-26:
> This issue is specific to SYNWIT screens units, it is recommended that users with this display compile the firmware with the option `HAS_GCODE_PREVIEW` disabled in the Configuration.h file.
> 
> Currently the control board does not have the ability to know what screen unit is attached so it is impossible to disable that feature automatically when that display is present. But it would be possible to add a check box in Advanced Settings to allow to disable all the confirmation and preview functionality.

Thanks for your help and look forward to the new function~

mriscoc, 2022-12-26:
I wonder if a correctly formatted G-code with preview information could cause the same bad behavior. I know that SYNWIT can't show the preview but maybe can show the print information.
I expect that you see something as:
[image]
Try to use this file and report if it also turns the display white:
[SimpleCuraTest.zip](https://github.com/mriscoc/Ender3V2S1/files/10304502/SimpleCuraTest.zip)

tomjschwanke, 2022-12-26:
Your testfile hangs my display immediately on the latest prebuilt release. In my own build without preview support it doesn't hang.

mriscoc, 2022-12-27:
> Your testfile hangs my display immediately on the latest prebuilt release, it doesn't turn white, just becomes unresponsive. In my own build without preview support it doesn't hang.

Thanks for the reply. So SYNWIT is totally incompatible with QR and Preview.

tomjschwanke, 2022-12-27:
I wonder if it's just the QR code / image. Maybe just the print info text could still work...

Enderprint, 2023-01-11:
and on the screen DACAL view can be implemented g coda very cool feature sorry with the screen no luck you are very good fellows for doing this

mriscoc, 2023-03-13:
Latest firmware allows to disable preview feature in Advanced Settings menu, that it fixes this issue.

closed 2023-03-13 by mriscoc

mriscoc, 2023-06-11:
SYNWIT/VIEWE displays had a problem with the preview feature in previous firmware versions, the screen turns to white/blank. In the latest version (Universal MAY 2023 Edition), the QR code feature was removed to improve compatibility with this display unit. Some users can even see a preview of the thumbnails.