mriscoc/Ender3V2S1 #324: When editing mesh, probing Z reference goes down instead of up

classicrocker883, 2022-08-27:
found another bug.

so when you select to edit the mesh in the mesh leveling settings, and then pick a point to find the Z reference, the nozzle/hotend moves down in the Z axis instead of going up.

this is the opposite of what you want so it doesn't accidentally scratch the bed.

mriscoc, 2022-08-27:
Good catch! Fixed in the next version.
```cpp
void BedLevelToolsClass::ProbeXY() {
  const uint16_t Clear = Z_CLEARANCE_DEPLOY_PROBE;
  sprintf_P(cmd, PSTR("G0Z%d\nG30X%sY%s"),
    Clear,
    dtostrf(bedlevel.get_mesh_x(BedLevelTools.mesh_x), 1, 2, str_1),
    dtostrf(bedlevel.get_mesh_y(BedLevelTools.mesh_y), 1, 2, str_2)
  );
  gcode.process_subcommands_now(cmd);
}
```

closed 2022-08-27 by mriscoc

classicrocker883, 2022-08-28:
how do you fix it so fast? do you know what the code should be or how do you solve it?

mriscoc, 2022-08-28:
> how do you fix it so fast? do you know what the code should be or how do you solve it?

The correct code is posted in my previous comment.

classicrocker883, 2022-08-28:
> > how do you fix it so fast? do you know what the code should be or how do you solve it?
> 
> The correct code is posted in my previous comment.

I know, I mean how do you know that is the correct fix? do you know computer programming languages very well?

mriscoc, 2022-08-28:
I am the programmer of Professional Firmware and its ProUI interface.

classicrocker883, 2022-08-29:
its good you know exactly what to do to fix this is a great firmware

classicrocker883, 2022-08-29:
the line should read

` const uint16_t Clear = Z_CLEARANCE_DEPLOY_PROBE + Z_HOMING_HEIGHT;`

because if it goes to just the z_clearance_deploy_probe height, and if there is a -Z offset value, it moves down. 

also in probe.h -> line 164

```
  static void move_z_after_homing() {
    #if ALL(DWIN_LCD_PROUI, INDIVIDUAL_AXIS_HOMING_SUBMENU, HAS_MESH)
      do_z_clearance(HMI_data.z_after_homing, true);
```
should it read as this? with HAS_MESH instead of MESH_BED_LEVELING ??

mriscoc, 2022-08-29:
Before editing the mesh or doing a bed tramming you should remove anything from the bed.