mriscoc/Ender3V2S1 #328: Display artifacts

Madmike-1337, 2022-08-28:
**Describe the bug**
There are odd text fragments appearing on the display during a print.

**To Reproduce**
Steps to reproduce the behavior:
1. Print any file
2. Observe Screen
3. See error

**Expected behavior**
The display to appear normally.

Photos Attached

**Version (please complete the following information):**
Ender3V2-422-BLTUBL-20220816.bin

**Additional context**
I am not 100% certain, but it looks like something may be coming from M73 and displaying off the screen. 

[image]
[image]

Madmike-1337, 2022-08-28:
This also appears to be intermittent. I am printing the same file again (acceleration test)  and now there is no artifact. I have been seeing this for several of the prints I have done since updating from the previous version.

mriscoc, 2022-08-28:
Those artifacts are caused by interference in the display communication. It can be mitigated shortering the display cable, or with a ferrite ring, but that will not eliminate it totally.
It is a knowed issue. You can clear them by turn on/off the backlight display in the tune menu.

closed 2022-08-28 by mriscoc