mriscoc/Ender3V2S1 #331: Octoprint fails with SKR Mini E3 V3

RoP09, 2022-08-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start print with Octoprint

**Expected behavior**
Print is started

**Log**
```
Recv:  T:191.90 /190.00 B:60.04 /60.00 @:8 B@:7
Recv:  T:191.74 /190.00 B:60.04 /60.00 @:12 B@:5 W:2
Recv: echo:busy: processing
Recv:  T:191.65 /190.00 B:60.03 /60.00 @:14 B@:7 W:1
Recv:  T:191.51 /190.00 B:60.01 /60.00 @:17 B@:17
Recv:  T:191.46 /190.00 B:59.90 /60.00 @:18 B@:52 W:0
Recv: echo:busy: processing
Recv: ok
Send: N12 M82*42
Recv: ok
Send: N13 M501*21
Recv: echo:M87 stored settings retrieved (732 bytes; crc 5679)
Recv: //action:notification Stored settings retrieved
Recv: Unified Bed Leveling System v1.01 active
Recv: Mesh loaded from slot 0
Recv: Mesh 0 loaded from storage.
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E130.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z20.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.10
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S1 Z2.00 ; Leveling ON
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: echo:Active Mesh Slot 0
Recv: echo:EEPROM can hold 19 meshes.
Recv: 
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H195.00 B60.00 F128
Recv: echo:  M145 S1 H230.00 B80.00 F128
Recv: echo:  M145 S2 H240.00 B60.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P22.89 I1.87 D70.18
Recv: echo:; Bed PID:
Recv: echo:  M304 P462.10 I85.47 D624.59
Recv: echo:; LCD Brightness:
Recv: echo:  M256 B150
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Retract (S<length> F<feedrate> Z<lift>):
Recv: echo:  M207 S5.00 W13.00 F2400.00 Z0.20
Recv: echo:; Recover (S<length> F<feedrate>):
Recv: echo:  M208 S0.00 W0.00 F2400.00
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-1.00 Y-45.80 Z-3.79 ; (mm)
Recv: echo:; Stepper driver current:
[... 139 lines cut ...]
Recv:  T:189.92 /190.00 B:59.99 /60.00 @:55 B@:20
Recv:  T:190.01 /190.00 B:59.95 /60.00 @:53 B@:34
Recv:  T:189.92 /190.00 B:59.97 /60.00 @:55 B@:28
Recv:  T:190.01 /190.00 B:60.00 /60.00 @:53 B@:19
Recv:  T:189.93 /190.00 B:60.00 /60.00 @:55 B@:20
Recv:  T:189.87 /190.00 B:60.05 /60.00 @:56 B@:5
Recv:  T:190.05 /190.00 B:59.94 /60.00 @:52 B@:38
Recv:  T:190.08 /190.00 B:60.01 /60.00 @:52 B@:15
Recv:  T:189.98 /190.00 B:59.94 /60.00 @:54 B@:40
Recv:  T:189.99 /190.00 B:59.96 /60.00 @:54 B@:33
Recv:  T:189.99 /190.00 B:60.01 /60.00 @:54 B@:17
Recv:  T:189.92 /190.00 B:59.99 /60.00 @:55 B@:23
Recv:  T:189.96 /190.00 B:59.99 /60.00 @:54 B@:25
Recv:  T:189.95 /190.00 B:60.06 /60.00 @:54 B@:0
Recv:  T:189.89 /190.00 B:59.95 /60.00 @:56 B@:37
```

**Version (please complete the following information):**
Latest config from special version (self compiled):
Recv: FIRMWARE_NAME:Marlin 2.1.0.1 MRiscoC Ender3V2-SKRME3V3-BLTUBL-T13-LA, based on bugfix-2.1.x (Aug 29 2022 09:45:14) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2-SKRME3V3-BLTUBL-T13-LA EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

**Additional context**
Just switched from Creality 4.2.2 to SKR Mini E3 V3 - no other changes done, same cable used.
Octoprint connects normally, but serial connection fails when print is started (after heating finished).
Printing from SD Card works fine.

RoP09, 2022-08-29:
Seems to be related to the start GCODE I used to use with my 4.2.2 board. 
If I print a file from my 2nd printer (Tronxy Crux 1) which uses totally different start GCODE the print starts fine with Octoprint.
Somehow the FW crashes on one of the start GCODE commands which previously worked on the 4.2.2 board.

RoP09, 2022-08-29:
Update: 
It is the M501; Read values from EEPROM - command that I have as first GCODE command in Cura. If I leave it away the print starts fine. 
Seems the repsonse takes too long and Octoprint does not receive a response on the following commands which leads to this serial error.

mriscoc, 2022-08-29:
Why you don't use 250000 baud?

RoP09, 2022-08-29:
Tried  both, same behaviour (just switched to the slower baudrate to rule at that this might be the root cause).

mriscoc, 2022-08-29:
Can you issue M501 manually using the console? because in my 4.2.2 I'm receiving the correct final ok without problems.

RoP09, 2022-08-30:
Immediately leads to a disconnect on Octoprint, but the echo ok is there:

[image]

RoP09, 2022-08-30:
Also I noticed that even 115K baud on the printer is not checked, Octoprint connects with 115K if set to AUTO. I can only use 250K if I force it in the baudrate settings in Octoprint.

mriscoc, 2022-08-30:
That's weird, I can't replicate that on my Ender3V2 and Octoprint. 
The firmware boots in 250k mode and it changes later to 115k if the option was enabled, this is the current supported configuration. If you force the configuration file to 115k, then it will boot at that speed, and only will change to 250k if you manually enable and then disable the 115k option.

RoP09, 2022-08-30:
no idea, in the FW I see the define is set to 250000 still...

mriscoc, 2022-08-30:
So, if in the Configuration.h file is set `#define BAUDRATE 250000` Octoprint connects to the printer at 115K?

RoP09, 2022-08-31:
Yes exactly. Seems Octoprint can connect at 115k and 250k when the printer is set to 250k but prefers to do in 115k.

mriscoc, 2022-08-31:
We need to have more confirmations about this issue, I received a report from another user that the SKR is connecting at 250k without issues with Octoprint installed in a Raspberry pi 4B.

RoP09, 2022-08-31:
Today I could not reproduce the behaviour, if I set Octoprint to AUTO baudrate, it connects now at 250k if the printer is set to 250k.
However, if I set Octoprint to 115k, it will also connect at 115k even if the printer is set to 250k...

michealmueller, 2022-09-20:
> Yes exactly. Seems Octoprint can connect at 115k and 250k when the printer is set to 250k but prefers to do in 115k.

I can confirm octoprint will default to 115k unless octoprint is forced to use 250k I also have an skrme3v3

mriscoc, 2022-10-04:
Please test with latest version.

closed 2022-10-04 by mriscoc