mriscoc/Ender3V2S1 #334: [BUG?] SKR Mini E3 V3 crashes if 5th stepper is added in UART mode

RoP09, 2022-08-30:
**Describe the bug**
The SKR Mini E3 V3 crashes if 5th stepper is added in UART mode.

Configuring the 5th TMC2209 in standalone mode works fine using the same pins, but once I set it to standalone the FW reboots endlessly after startup (immediately after the main control screen becomes visible, cannot rotate the knob or see the FW name scrolling into the view yet).

**To Reproduce**
Configuration used:
[image]

**Expected behavior**
TMC 2209 also should work in UART mode

 
**Version (please complete the following information):**
Latest config from special version (self compiled):
Recv: FIRMWARE_NAME:Marlin 2.1.0.1 MRiscoC Ender3V2-SKRME3V3-BLTUBL-T13-LA, based on bugfix-2.1.x (Aug 29 2022 09:45:14) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2-SKRME3V3-BLTUBL-T13-LA EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

**Additional context**
Not sure if this is a bug or wrong configuration, as I cannot find any software serial settings for that board (not supported at all?). All other steppes use hardware serial it seems.

mriscoc, 2022-08-30:
Please, first test your configuration with the latest Bugfix source code from Marlin repository. That could be a problem with Marlin core.

RoP09, 2022-08-31:
You mean the one from BTT here? 
https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1

Or official Marlin?

mriscoc, 2022-08-31:
Oficial Marlin Bugfix 2.1.x with the same configuration files.

RoP09, 2022-08-31:
Seems I broke my board now with this :(

Copied over the following files to the official Marlin 2.1.x bugfix and compiled fine, but after flashing the screen stays black:

[image]

After flashing back to your FW, the screen also stays black now .. any suggestions?

mriscoc, 2022-08-31:
Ok, thanks for trying.
Your issue seems to be something that needs to be fixed on the Marlin side, I don't have a SKR to try to fix.

mriscoc, 2022-08-31:
Because it is not a problem with the Professional firmware with default configuration, I'm closing this issue by now, it will be re-opened if this configuration is confirmed to works with the current Marling bugfix branch.

RoP09, 2022-08-31:
Also thought the pins. But the behavior is the same If i use other pins in the board.

RoP09, 2022-09-05:
Did another try with setting  the configuration to **MSerial5** instead of the RX/TX pin config. Now the printer does not reboot, but shows a message "**TMC_CONNECTION_ERROR**" when booted up.

With manual control I can even move the 2nd Z stepper now, but it seems in this situation it somehow works half speed of the 1st stepper, so I went back to step/dir mode.

Seems something is still wrong either in wiring or in the configuration of those TMC2209s in UART mode.

Any ideas?

mriscoc, 2022-09-05:
https://3dprintingspace.com/u/sts41
https://marlinfw.org/docs/gcode/M122.html
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/Wiring%20diagram/SKR%20V1.3-TMC2208-UART-Mode.png

Also, don't forget to disable the +V if you connect your board trough USB.

RoP09, 2022-09-05:
Thanks! Will give that M122 a try, forgot about it ...

The V+ I think is not an issue on the SKR boards, there is a jumper to set USB voltage supply which is not set.

RoP09, 2022-09-05:
[image]

mriscoc, 2022-09-06:
It seems like a problem with the jumpers/connections of that driver.

RoP09, 2022-09-06:
Swapped Z2 with Y now and connected Y to the external driver in standalone, only solution that works it seems...

mriscoc, 2022-09-06:
Currently Linear Advance works also with drivers in standalone mode.