mriscoc/Ender3V2S1 #335: Adding the option change Mesh Viewer

classicrocker883, 2022-08-31:
### Adding Mesh Viewer - change how the mesh is viewed
enables both options on how the mesh is viewed
with `USE_UBL_VIEWER` enabled, this gives a check box to alternate type of view for mesh

</BR>

</BR>

adding Menu Item
FILE-->   dwin.cpp
   under Draw AdvancedSettings Menu
   

      ```MENU_ITEM(ICON_MeshViewer, MSG_MESH_VIEW, onDrawSubMenu, DWIN_MeshViewer);  
      #if ENABLED(USE_UBL_VIEWER)
        EDIT_ITEM_F(ICON_PrintSize, "Change Mesh Viewer", onDrawChkbMenu, SetViewMesh, &BedLevelTools.view_mesh); 
      #endif```
</BR>

</BR>
      
      
//the function      
FILE-->   meshviewer.cpp

```
void MeshViewerClass::Draw(bool withsave /*= false*/) {
  Title.ShowCaption(GET_TEXT_F(MSG_MESH_VIEWER));
  #if ENABLED(USE_UBL_VIEWER)
  if(BedLevelTools.view_mesh) {
    DWINUI::ClearMainArea();
    BedLevelTools.viewer_print_value = true;
    BedLevelTools.Draw_Bed_Mesh(-1, 1, 8, 10 + TITLE_HEIGHT);}
  else
    DrawMesh(bedlevel.z_values, GRID_MAX_POINTS_X, GRID_MAX_POINTS_Y);
  #endif
  if (withsave) {
    DWINUI::Draw_Button(BTN_Save, 26, 305);
    DWINUI::Draw_Button(BTN_Continue, 146, 305);
    Draw_Select_Highlight(HMI_flag.select_flag, 305);
  }
  else
    DWINUI::Draw_Button(BTN_Continue, 86, 305);

  #if ENABLED(USE_UBL_VIEWER)
    if(BedLevelTools.view_mesh) {
      BedLevelTools.Set_Mesh_Viewer_Status();}
    else {
    char str_1[6], str_2[6] = "";
    ui.status_printf(0, F("Mesh minZ: %s, maxZ: %s"),
      dtostrf(min, 1, 2, str_1),
      dtostrf(max, 1, 2, str_2)
    );}
  #endif
```

 
</BR>

</BR>
  
  
  
  
FILE--> settings.cpp

```
  #if ENABLED(USE_UBL_VIEWER)
    bool view_mesh;
  #endif
```
//under MarlinSettings::save()

```
  #if ENABLED(USE_UBL_VIEWER)
    EEPROM_WRITE(BedLevelTools.view_mesh);
  #endif
```    
//under MarlinSettings:_load()

```
  #if ENABLED(USE_UBL_VIEWER)
    _FIELD_TEST(view_mesh);
    EEPROM_READ(BedLevelTools.view_mesh);
  #endif      
```
//under MarlinSettings:reset() 
```
  #if ENABLED(USE_UBL_VIEWER)
    BedLevelTools.view_mesh = ENABLED(USE_UBL_VIEWER); //added mesh viewer option
  #endif 
```
</BR>

</BR>

FILE--> bedlevel_tools.cpp
 
//under
```BedLevelToolsClass BedLevelTools
 
  #if ENABLED(USE_UBL_VIEWER)
    static bool viewer_asymmetric_range;
    static bool viewer_print_value;
    static bool view_mesh;
  #endif
```
  
</BR>

</BR> 
  
FILE--> bedlevel_tools.h

```
  class BedLevelToolsClass {
public:
  #if ENABLED(USE_UBL_VIEWER)
    static bool viewer_asymmetric_range;
    static bool viewer_print_value;
    static bool view_mesh;
  #endif
```
  

[image]
[image]

mriscoc, 2022-08-31:
Be aware that this duplicates the code of the viewer, that is why I leave it only as a user's selectable at compile time.
Currently the RCT6 devices are close to the limit of the program's memory.

classicrocker883, 2022-08-31:
i see, maybe that can be tweaked so it limit the memory

classicrocker883, 2022-08-31:
using the RET6...

Without `USE_UBL_VIEWER` enabled
RAM:   [==        ]  23.3% (used 15300 bytes from 65536 bytes)
Flash: [====      ]  41.7% (used 218444 bytes from 524288 bytes)

With `USE_UBL_VIEWER` enabled
RAM:   [==        ]  23.4% (used 15304 bytes from 65536 bytes)
Flash: [====      ]  41.9% (used 219876 bytes from 524288 bytes)