mriscoc/Ender3V2S1 #344: Print menu doesn't appear if start gcode contains C108

paulorsv01, 2022-09-01:
**Describe the bug**

Found a bug when using C108 in my starter gcode to auto-press continue when the mesh result screen is being shown. If I don't use C108, the print menu appears normally. If I use C108, the print menu doesn't appear, but the print happens normally.

**To Reproduce**

Not sure if it happens on other versions, so these steps are for `Ender3V2-422-BLT-20220816` version.
1. Add C108 to starter gcode, right after G29.
2. Slice and try to print something.
3. It will stay on the main menu.

**Expected behavior**

To show the print menu.

**Screenshots**

https://user-images.githubusercontent.com/16693325/187953681-50c88d3d-ea83-40b5-ae6d-ad80b4bc5c55.mov

Both files are almost the same, with the difference being that one contains C108, and the other doesn't.

**Version (please complete the following information):**

Ender3V2-422-BLT-20220816

**Additional context**

My current start gcode for Cura is:

```gcode
M104 S130 ; Preheat nozzle to 130
M140 S{material_bed_temperature_layer_0} ; Start heating bed without waiting
M105 ; Report temperature
M190 S65 ; Wait for bed to reach temp before proceeding
M82 ;absolute extrusion mode

M501; Read values from EEPROM
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z20.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; ABL.
C108 ; Continue

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y0.1 Z3.0 F5000.0 ; Move to start position
M104 S{material_print_temperature_layer_0} ; Start heating hot end without waiting
M105 ; Report temperature
M109 S{material_print_temperature_layer_0} ; Wait for hot end to reach temp before proceeding
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

mriscoc, 2022-09-01:
Hi @paulorsv01, thank you for your report, that is indeed a bug caused by the new g-code C10 for the configuration file. I will to fix it in the next release.

classicrocker883, 2022-09-02:
is there a way to have a highlighted box around the Continue button? as it is selected, but it seems like it should have a box around it

closed 2022-10-02 by mriscoc

DT61, 2022-12-12:
Hi , Great bed leveling tools/concepts , but one issue hot end cools during leveling process before printing. Ender V2 4.2.2 BLTouch.
Like the image view on GUI nice touch. 
Looking forward to your next release.

commit afa9df35 (2022-10-19): Professional Firmware - Austral Spring version 20221008
- Shortcut option for turning off the backlight in toolbar
- Filament run-out sensor enable/disable in tune menu
- Enhanced Host printing support
- Better C10 support
- Axis value in dash board shows actual position of the axes, including Z leveling
- Raise Z to Probe Clearance position before of a Probing Z Reference
- New ProUI Info box
- Fix #344
- Fix mesh slot load message bug
- Enable support for Creality Slicer G-code file with embedded thumbnail for Ender3V2 Neo
- Implements a user option for center the menu title by defining TITL [... cut]