mriscoc/Ender3V2S1 #347: Multiple probing ingores a value of 1

AgenB, 2022-09-01:
**Describe the bug**
When I try to change the default multiple probing amount from 2 (which results in 3 probes per point) to 1, it changes to 0 by itself, resulting in just 1 probe per point, can't set it to 2.

**To Reproduce**
1. Go to Advanced settings -> Probe settings
2. Set Multiple Probing to 1, it changes to 0 after you reopen the menu.

**Version:**
Ender 3 v2 4.2.2 BLT UBL 20220816

mriscoc, 2022-09-01:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/319

closed 2022-09-01 by mriscoc

classicrocker883, 2022-09-03:
@AgenB 

(line ~2290 in file proui/dwin.cpp)

void ApplyProbeMultiple() { PRO_data.multiple_probing = (MenuData.Value > 0) ? MenuData.Value : 0; }

change the value to be greater than 0 rather than 1.

so if you later choose muliprobing in the menu, a zero value would make 1 time slow probe.
value of 1 will make 2 probes, 1 fast and 1 slow.
each one adds one. it is basically ADDING probes rather than MULTIPLE probes

mriscoc, 2022-09-03:
> @AgenB
>
> (line ~2290 in file proui/dwin.cpp)
>
> void ApplyProbeMultiple() { PRO_data.multiple_probing = (MenuData.Value > 0) ? MenuData.Value : 0; }
>
> change the value to be greater than 0 rather than [...]

Be aware that is not correct, PRO_data.multiple_probing must be forced to 0 for any value lower than 2.