mriscoc/Ender3V2S1 #348: Filament missing warning

pacocatech, 2022-09-02:
Hello I'm using the versions you released for ender 3 v2 board 4.2.2

releases: 20220816 / 20220716 / 20220419 BLT / MM

In all versions at some point the printing stops and it keeps saying that filament is missing, is there a known bug for this?
When I have the original creality FW this process does not happen.

[image]

mriscoc, 2022-09-02:
https://github.com/mriscoc/Ender3V2S1/issues/227

pacocatech, 2022-09-02:
I've already tested with the LOW / HIGH setting and I still get the filament change warning.
Even if you disable the sensor or the warning continues

mriscoc, 2022-09-02:
Do you have a filament runout sensor?

pacocatech, 2022-09-02:
yes

mriscoc, 2022-09-03:
Ok, in Advanced Settings there is a tool: End Stop Diagnostic, Runout must be in TRIGGERED status when there is filament inside.

pacocatech, 2022-09-03:
I will check

pacocatech, 2022-09-03:
@mriscoc 
With filament:
[image]

without filament:

[image]

mriscoc, 2022-09-03:
That is correct, so if you got a pause inspect your G-code files for pause commands as M25, M125 or M600.

pacocatech, 2022-09-04:
This is my gcode 
[Teil1_2h51m_036mm_240C_PETG_ENDER3V2.zip](https://github.com/mriscoc/Ender3V2S1/files/9484916/Teil1_2h51m_036mm_240C_PETG_ENDER3V2.zip)

doesn't have any pause gcode.

mriscoc, 2022-09-04:
Check your filament sensor connections. This is not a firmware issue.

closed 2022-09-04 by mriscoc

pacocatech, 2022-09-05:
then!
my sensor is new... with the original creality firmware i don't have this problem.
If there's any more tests I can do to confirm that there isn't a bug in the firmware version just let me know, I'd really like to understand what's going on to use your version.

mriscoc, 2022-09-05:
Your pictures show the firmware can recognize your filament sensor correctly, I also have a V2 with 4.2.2 with filament sensor and don't have any problem. If you check your G-code and do not have any pause command, I can't figure out what is happening, but a connection problem. With Creality firmware you will not have this problem because that firmware doesn't support filament sensor support and ignores it.

pacocatech, 2022-09-05:
I'm going to compile a new version of marlin, with the latest version available in the community and I'll test it, I'm sure the problem is not the sensor and it's not in the gcode either. Thanks for listening