mriscoc/Ender3V2S1 #356: Toolbar Icon alternates

classicrocker883, 2022-09-09:
Sometimes after compiling and flashing, the 2nd item Icon in the Toolbar doesn't stay what it is supposed to be.

It is labeled Tramming Wizard and supposed to be `ICON_SetHome`. The Icon becomes PLA (`ICON_PLAPreheat`). This is intermediate and happens sometimes with new firmware compile and flash.

i find in the code this error attributed to TBItemA[]
<br>
[image]
</br>

i notice that the PLAPreheat icon is the next number in sequence in FILE _dwin_set.h_

```
#define ICON_Homing             29
#define ICON_SetHome            30
#define ICON_PLAPreheat         31
```

sometimes I can temporarily fix this by changing the Tramming Wizard Icon to `ICON_Homing`, and it becomes the correct icon when flashed

classicrocker883, 2022-09-10:
the error remains after the change, and after trying a brand new download of the files

classicrocker883, 2022-09-10:
the only type of fix it suggests is to edit the IncludePath. it suggested i should have vcpkg installed, which i do. perhaps there is a library it is referring to that I do not have.

classicrocker883, 2022-09-10:
[image]

as you can see, it cuts off here.

<br>

[image]

and this is from the original source

classicrocker883, 2022-09-10:
[image]

it is saying error-type as if the struct / class or typedef is incorrect

classicrocker883, 2022-09-10:
[image]

after having Arduino.h in the include path, more errors pop up

classicrocker883, 2022-09-10:
i think i fixed it.
i added `#include "../HAL/shared/progmem.h"`
to FILE multi_language.h

mriscoc, 2022-09-10:
Maybe you have an error in your configuration files.

mriscoc, 2022-10-04:
It is not possible to replicate.

closed 2022-10-04 by mriscoc