mriscoc/Ender3V2S1 #357: missing progress bar

Tjjohnston, 2022-09-10:
I cannot see the progress bar to any degree while a print is going. I saw in the preview images there was a rainbow looking progress bar, though I have nothing at all that shows me any information about the print, nothing changes at all from when a print is going and when a print is not going. I realize that is a feature to some degree, but I would like to know the print progress and such. 

I realize this is more than likely a user error and possibly I am not seeing a setting or something to turn on. Though it has been a few hours of fiddling and hunting and I have yet to have any changes. 

Also, I greatly appreciate your work here. 3D printing has been a nightmare for me trying to get things to work well, but this firmware is truly amazing and simple. Thank you. 

**To Reproduce**
Steps to reproduce the behavior:
1. Just run a print normally

**Expected behavior**
The progress bar to show up while a print is going.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Ender3V2-422-BLTUBL-20220816

IMPORTANT:
I would also note, while trying to get my screen firmware changed I ran into some serious "black screen" issues. I say serious because no matter what I did no change would take effect and the screen was bricked. I went through all the steps to rename the bin and all. I even reverted back to the original firmware. Nothing worked. It wasn't until I ran across a reddit post about this exact issue. Apparently some recent Ender 3 v2 printers have a strange set up for reverting the screen firmware. 
- No matter what version of printer you have (I know  you specified that you have to use either the DWIN or private folder depending on your version)
- With these certain printers, you have to use DWIN, but then you have to rename the file "private"
- I really hope you share this information, I was literally about to buy another motherboard because I could not find a solution.

Tjjohnston, 2022-09-10:
I also saw that I have no control to change the current prints speed, or any way to stop the print. I had to turn off the printer to stop the print because the speed was too high. I feel like this is indeed a bug.

classicrocker883, 2022-09-10:
[image]

did u use the "stock dwin set"?
make sure to uncomment this line

u may have to install the correct DWIN set into the LCD.

Tjjohnston, 2022-09-10:
> [image]
> 
> did u use the "stock dwin set"? make sure to uncomment this line
> 
> u may have to install the correct DWIN set into the LCD.

I just used the precompiled firmware. "Ender3V2-422-BLTUBL-20220816"
I will have to look and see how to try and compile my own for mriscoc. Honestly I am little unsure of that kind of thing, but I did make it work on Marlin. I will do some research and see if I can find that. 
My apologies if I am misunderstanding your comment, I really appreciate the quick response!

mriscoc, 2022-09-10:
Are you printing from Octoprint of from SD card?
For Octoprint follow this guide: https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

Tjjohnston, 2022-09-10:
> This firmware will works with stock firmware display, you don't need to update it.

I am printing from an SD card. 
Ok, so I should flash the display to the stock firmware? I flashed with the Giadej set, everything looks like it shows in your photos, I am just missing the "control menu icons"

Thank you so much for your help

mriscoc, 2022-09-10:
You can use the Giadej icon set published here.

Maybe your G-code has a C108? In the latest version there is a bug: https://github.com/mriscoc/Ender3V2S1/issues/344

Tjjohnston, 2022-09-10:
Hmm, the Giadej is the icon set I was using when I did not have the control menu. 
I will re insert my SD card with the recommended Gcode settings with cura, maybe that will ensure nothing I had in my Gcode was causing issue. I will let you know if that fixes anything.

Tjjohnston, 2022-09-10:
[image]

Tjjohnston, 2022-09-10:
After reflashing the Giadej I am still in the same situation with no control icons, I also refreshed my Gcode and made sure C108 was not included, though it wasn't before. 

The image I posted is when a gcode file is selected to print, and my printer is preheating, though the screen never changes from this for the duration of the print at all.

mriscoc, 2022-09-10:
Can you post your G-code file please?

limamg, 2022-09-11:
if the printing is being ugly by the sdcard the progress bar always appears.
If the printing is being done by octoprint, it is necessary to configure it as explained on the website
[https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#octoprint-gcode-scripts](url)

mriscoc, 2022-10-04:
No more user feedback

closed 2022-10-04 by mriscoc