mriscoc/Ender3V2S1 #359: screen shows distorted numbers

thesebastianf, 2022-09-10:
sorry - no clue how to reproduce.
installed firmware is Ender3 V2/S1 Professional Firmware 20220816
on a Ender 3 V2 4.2.2 BLTOUCH

on a very long print - after 12 hours the display started to show numbers in big letters randomly on the screen
see screenshot top and down area

[image]

mriscoc, 2022-09-10:
That screen glitch is due to interference in the cable between the control board and the screen, sadly that can not be fixed by firmware, some users reports that using a ferrite ring (choke) can help with that issue. Also, check that cable and re-route it.

closed 2022-09-10 by mriscoc

lagnat, 2022-09-28:
I had this happen yesterday right after installing this firmware.  Never once happened before.  Just my $.02.

mriscoc, 2022-09-29:
> I had this happen yesterday right after installing this firmware. Never once happened before. Just my $.02.

Stock firmware doesn't send less information to the screen while it is printing.

lagnat, 2022-09-29:
Less information?  Never had the problem with Jyers either.  I'll try the ferrite.  What about a shielded cable?

mriscoc, 2022-09-29:
> Less information? Never had the problem with Jyers either. I'll try the ferrite. What about a shielded cable?

This firmware may not be compatible with your hardware, I recommend going back to a functional firmware.

lagnat, 2022-09-29:
Ender3V2, v4.2.2 board

mriscoc, 2022-12-19:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/328