mriscoc/Ender3V2S1 #360: Extruder motor ignoring maximum speed printer startup

AsterDaishi, 2022-09-10:
I came across this bug while calibrating my printer's e-step. I'm currently running the firmware Ender3v2-422-BLTUBL-MPC.

The behaviour as I understand it seems to cause the first extruder movement to execute at a much higher speed than normal: in my case, it ran the equivalent of 100mm of filament in around 2 seconds or so. Any subsequent extruder movmement after the first follow the max speed limit as expected. 

The steps that lead to the bug are the following: 

- turn on printer
- preheat pla (or anything else, as long as the nozzle is above the minimum extrusion temp)
- from the menu, prepare > move axis > move extruder - 100mm

As previously mentioned, this will happen the first time only. Any other extruder movement will work properly until the printer is turned off and on again and the steps repeated.

classicrocker883, 2022-09-12:
i think this has to do with loading the filament. i do recall this does happen at first. so it thinks the filament isnt really loaded and it makes the filament go fast through the tube (if its a bowden) and slows onces it reached a certain parameter - maybe time, or distance (steps), not sure.

AsterDaishi, 2022-09-12:
That's possible, but it doesn't seem to be distance or time based at all. As a matter of fact, in order to complete the e-step calibration I needed to do, I would make sure that the first extrude movement after booting the printer would be 0.1mm. this would ensure that the following 100mm movmenent would be executed at the correct speed.

mriscoc, 2022-09-12:
Could be the default feedrate. I will check this for fix it in the new release version.

mriscoc, 2022-09-12:
Sorry, I was no able to replicate this bug with firmware version: [Experimental-Ender3V2-422-BLTUBL-MPC-20220816.bin](https://github.com/mriscoc/Special_Configurations/releases/download/MPC/Experimental-Ender3V2-422-BLTUBL-MPC-20220816.bin)
Steps:
1- Firmware flashed and defaults loaded
2- Turn off / Turn on printer
3- Prepare / Preheat PLA (wait for target temperature)
4- Prepare / Move Axis / Move Extruder 100 mm
The extruder moves at the programmed speed of 1 mm/s

AsterDaishi, 2022-09-27:
Sorry about the late reply. That's odd, I could get that behaviour to happen consistently. The only further information I think I can add is that the 4.2.2 board was one of those with a STM32F103RCT6 controller on it and according to the letter marked on the SD card should have HR4988 stepper drivers. Don't know if this would be of any help.

Unfortunately I've recently upgraded my printer to a SKR board, so further testing with the previous board would be a bit problematic. Just for the sake of it, I tried to also replicate the issue on the new board/firmware combo (still using your firmware) but as expected everything behaved correctly.

mriscoc, 2022-10-04:
Thanks, I'm closing this issue.

closed 2022-10-04 by mriscoc