mriscoc/Ender3V2S1 #361: Manual mesh Y axis is mirrored

MaynardMann, 2022-09-12:
ender 3 s1 f4 chip 

as the title says Iv been trying to figure out why the manual mesh makes things so much worse for a week iv finally got it the y axis is mirrored same for x I think

classicrocker883, 2022-09-12:
in the Configuration.h file

line 1640
```
// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false  // Ender Configs
#define INVERT_Y_DIR false  // Ender Configs
#define INVERT_Z_DIR true 
```

set the False to true.

### OR what you can do

is swap two of the 4 wires going into the stepper motor. be sure that the printer is OFF when you do this.

and look up search which two wires to swap in the connector

mriscoc, 2022-09-12:
There are some issues with S1 F4 boards and autoleveling, please check this: https://github.com/mriscoc/Ender3V2S1/issues/195

Currently, the only solution is creating a mesh and adjusting it using manual editing.

closed 2022-09-12 by mriscoc