mriscoc/Ender3V2S1 #366: G34 Option in menu

tpups, 2022-09-12:
**Some UI like Marlin have an option to run G34 in the menu. In Marlin I believe it's under Motion > Z Auto Align**
It would be great if it could be added to Professional Firmware for those of us with dual z-axis + dual steppers.

**Additional context**
https://marlinfw.org/docs/gcode/G034-zsaa.html

mriscoc, 2022-09-12:
That might be a candidate for a custom menu, but you'll need to compile your own firmware version.

tpups, 2022-09-12:
Yeah that would be no problem. I was able to build Professional Firmware with G34 enabled, just would be nice to have the option of enabling it in the menus.

tpups, 2022-09-14:
Ah just found the custom menu options in configuration.adv. Didn't realize that might be an option. Attempting to add to Configuration Menu.

tpups, 2022-09-14:
Is it possible to add as a custom menu item using either CUSTOM_MENU_MAIN or CUSTOM_MENU_CONFIG in Configuration_adv? Where would those options appear in the UI?

mriscoc, 2022-09-14:
Of use a special configurarion g-code file as: https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files

dennisMe2, 2023-01-12:
Can someone point me to some docs to learn how to implement this? I'd really like it. 
For now I have a file on each SD card so I can run it before generating a mesh or tramming but its clunky!

VovyCZ, 2023-01-15:
I would also like this option in the firmware menu for the SKR MINI E3 V3.

sammcj, 2023-03-03:
Upgrading to the SKR Mini E3 V3 this weekend - this would be nice to have!

gsuresh2u, 2023-03-03:
SKR mini e3 v3.0 has dual stepper connectors. But they both driven by single driver
G34 option is available only for individual Z steppers
Not possible for SKR mini e3 v3

VovyCZ, 2023-03-03:
I know, but it dawned on me later. In the meantime I have upgraded to the Klipper and after some trials I am thinking of putting your firmware back into the Ender 3 V2 and selling it. As a replacement, I'm thinking about something with the Klipper, probably the RatRig V-Core 3.1. But this is offtopic...

closed 2023-05-14 by github-actions[bot]