mriscoc/Ender3V2S1 #373: [Feature Request] Disable QR Code information for people with DACAI screens.

thantik, 2022-09-15:
I've seen some customers confused, and/or a little perturbed seeing the QR code telling them how to generate a gcode preview when they either A) Don't care about setting it up, B) It's too intimidating for them to try, or C) They have a DACAI screen and can't anyhow.

I'd love a C-code added to disable this QR generation/info

It's possible for me to just generate the gcode previews anyhow, but I run a 3D printer repair shop, and my customers don't have that same option all of the time.  This is purely a feature request as a "nice to have", not something that fixes any actual "problem".

GT-Geo, 2022-09-15:
I second the request to be able to disable the QR code.

mriscoc, 2022-09-15:
Disable `#define HAS_GCODE_PREVIEW` in Configuration.h file

closed 2022-09-15 by mriscoc

thantik, 2022-09-19:
Is this just something that isn't worth the time?  Many of my customers I send here to upgrade their firmware and they don't know how to compile their own.  I guess it means more business for me, but the addition of a C-code/M-code, etc would ease that burden.  Would you accept a pull request for such a change if I end up having to hire a developer to do it?

mriscoc, 2022-09-20:
Because currently RCT6 devices almost fill their program memory I am only implementing features that are important to a large number of users. But there are also sponsored configurations: https://github.com/mriscoc/Special_Configurations
Currently DACAI screens cannot display the G-code preview but can display all the other information about the G-code, only the preview area will be empty. So, no QR code is shown.

[image]

thantik, 2022-09-21:
Thanks for your time and commitment to this project!  I should have hung onto that display from yesterday, as it was a DACAI screen - I had swapped it out for a DGUS some time ago; but this particular DACAI screen showed the QR code despite what you're informing me should happen.  I appreciate your knowledge on the subject anyhow, and I will try to figure out a less troublesome way of fixing it for everyone involved.  Thanks for the links to the sponsored configs.