mriscoc/Ender3V2S1 #380: Extruder purging excess filament after print resume

swachalitbharat, 2022-09-16:
**Extruder/Nozzle overflow after print resume **
First time posting here. Well to begin with i have had this bug now during 2 consecutive prints. The extruder motor is running at high speed after print resume. I am running the firmware package Ender3V2-422-BLT-20220816. While printing my favorite gridinfinity packs the power went off and after resume i got the usual power loss message so i confirmed to continue printing. The bed and nozzle heated up to desired temperatures and then nozzle went and parked at the parking position for cleaning. Once i cleaned the nozzle i selected to continue and then printer homed the x and y axis and went to last known z height. This is where it just started overflowing like crazy. i press pause it didnt pause, so i just went power off again. Resumed repeated the steps same thing. Resumed again this time asking the printer to pause and go in advanced pause mode. Sadly enough the printer homed on top of my print and went back to 0% at this point. Has happened in two prints and after last one i decided to post here.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print.
2. Power off mid print.
3. Switch on printer again
4. Select to continue print after power loss.
    4.1.(a)  Let the bed and nozzle heat
    4.1.(b) Clean the nozzle at park position and press continue.
    4.1.(c)  Watch the nozzle purge filament out like a paste squeezer.
                                                OR
    4.2.(a) Select 'Pause' and let the bed and nozzle heat up.
    4.2.(b) Printer will go in advanced parking and ask to purge or continue. At this step select continue.
    4.2.(c) Printer will go in parking again and ask you to clean nozzle and continue. Press continue.
    4.2.(d) Watch the print complete % go to 0 and print resume from begining after CR touch homes on the print

5. Based on the path taken in step 4 you either end up with printer purging filament on top of you print like crazy or in the latter case start the print from begining

**Expected behavior**
Print to resume from where it stopped and not purge out lotsa filament with extruder extruding at high speed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Ender3V2-422-BLT-20220816

**Additional context**
Using cura slicer to slice my print files. Printer has no issue as i can start prints and finish them without trouble unless power goes out.

mriscoc, 2022-09-16:
I had that report (4.1) before, it seems to be a bug not easily reproducible because it depends on the G-code file and where the power outage was produced. Recovery after power failure is not 100% reliable and how it works is managed by Marlin core. Could be interesting to try with another slicer to test if the bug is also present.

swachalitbharat, 2022-09-16:
I would like to add that I had filament runout sensor enabled without actually installing it. I ended up disabling it and the extruder overflow bug disappeared but then i ended up with x axis shift after print resume. I am actually heavily considering investing in a good ups. The print resumed from correct z height but had shifted +5mm maybe in the x direction. Could be another bug. I dont get poweroutages for more than 10-15 minutes unless some major trasnformer upgrades happen so i thought it would be ok without the UPS.
I think the bug is related to having runout sensor active but not installed. no extruder overflow issues anymore. Have checked 3 times but now getting that +4 or 5 mm x offset.
 Your firmware is amazing and I am very new to 3D printing and going from stock to this made my printer so much better and faster i think. So i do salute you and all the contributors for all the work put in. Hopefully can make a contribution through a properly running printer. Its amazing how fast you responded.

mriscoc, 2022-09-17:
Your X displacement could be due to the impossibility to reach the X parking position. Check your X max, and X parking position.
Check this: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

swachalitbharat, 2022-09-17:
> Your X displacement could be due to the impossibility to reach the X parking position. Verify your X max, and X parking position. Check this: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
 
I sliced it with creality slicer and had same issues. Used a calibration square for all print failure detections in my case.

I read through that link you sent. All settings were accurate in the precompiled bin file except the x offset. It was off approximately by -5mm and now i have no issues after power outage it seems. I hope that fixed it. My x offset came to be -46.5mm and y offset as -6.5mm. I used a ball point on a paper to measure it as accurately as possible. That calibration guide is gold. Wish i had read it before. Now I know more than I knew yesterday. Thank you. Appreciate the help.
 
 To sum it up. I fixed the overflow issue after print resume by disabling the filament runout checkbox in filament management. I havent checked if the problem is persistent after installing a runout sensor. 
 The x offset after print resume was fixed by changing Xoffset and Y offset values to accurately represent my printer's configuration.

closed 2022-09-17 by swachalitbharat