mriscoc/Ender3V2S1 #389: Zits when printing from Octoprint

JonSnow88, 2022-09-22:
**Describe the bug**
When printing from Octorprint using Cura vase mode, I get zits/blobs during my print. The head pauses for a few seconds and then resumes, as far as i can tell this only occurs when using vase mode. I've tried using Arc Welder to see if this would help and it does but does not resolve the issue. The zits do not show up when printing from the SD Card, only from Octoprint. 

Chatting on the Octoprint Discord it was recommended to increase the buffer size to resolve this issue.

What is the current buffer size of August release and can this be increased?

ZipperZ, 2022-09-23:
For me this was present not only in vase mode. But I have assumed that it was Octoprint issue, because at the same time as I have changed the FW I also tried some plugins. For now I am printing from SD and planning to reinstall Octo. 
Tagging along to see if buffer size will/would help

CorDawg, 2022-09-24:
So I think this issue is related to Octoprint plugin "Resource Monitor". I have since disabled this plugin and have had no issues since.

JonSnow88, 2022-09-25:
> I think this issue is related to Octoprint plugin "Resource Monitor". I have since disabled this plugin and have had no issues.
> 
> [Renaud11232/OctoPrint-Resource-Monitor#37](https://github.com/Renaud11232/OctoPrint-Resource-Monitor/issues/37)

Nope, I had Resource Monitor plugin disabled for a while do to the zits. I uninstalled Resource Monitor a few days ago as well but vase mode still has zits.

I've run various test from, reinstalling octopi, stopped using the web interface or apps while printing, using octoprint in safe mode. Nothing I have tried have removed the zits when in vase mode.

liuxyc, 2022-09-25:
double BUFSIZE,TX_BUFFER_SIZE,RX_BUFFER_SIZE,  it works for me

JonSnow88, 2022-09-25:
> double BUFSIZE,TX_BUFFER_SIZE,RX_BUFFER_SIZE, it works for me

How do you do this?

liuxyc, 2022-09-25:
> 

1. edit  BUF defines in  Marlin\Configuration_adv.h  file
2. compile firemare

mriscoc, 2022-09-25:
The coming version will have buffers increased.

mriscoc, 2022-10-02:
Seems to be fixed in the latest release

closed 2022-10-02 by mriscoc