mriscoc/Ender3V2S1 #410: how to save my current machine settings Before flashing  ?

CaVO32, 2022-10-03:
each time i try to update my firmware it sets to default all my machine settings.... how do i save these settings before i flash or upgrade. 
i know your will say to use the [Configuration Files](https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files) for setting my tuned settings AFTER i have upgraded the firmware but i want to save these settings  B E F O R E   upgrading ?

mriscoc, 2022-10-03:
Use [M503 ](https://marlinfw.org/docs/gcode/M503.html) to get your current settings.

closed 2022-10-03 by mriscoc

CaVO32, 2022-10-05:
so you are saying that i need to use other software like octoprint or similar to do this..??

limamg, 2022-10-05:
> so you are saying that i need to use other software like octoprint or similar to do this..??

in Octoprint go to the terminal and type the command, it will display all the settings there you copy and generate the file to save your printer configuration

mriscoc, 2022-10-05:
Also using a PC and installing Pronterface