mriscoc/Ender3V2S1 #428: Error MAXTEMP

wassereimer86, 2022-10-08:
**Describe the bug**
I used the previous version from August fpr my S1 Pro (F4) with the special configuration https://github.com/mriscoc/Special_Configurations/releases/tag/T13 without any problems. After doing the same changes to the newest version and after the firmware update is done the printer reboots and begins to beep fast with the message "Nozzle or Bed temperature is too high" in the display and very quick after this it's beeping continuously with the message "Printer killed! Err: MAXTEMP Turn off the printer"

**To Reproduce**
Steps to reproduce the behavior:
1. Compile with special configuration

**Expected behavior**
Working fine

mriscoc, 2022-10-11:
Could you test with the latest sources? try removing some features to make more program memory space.

wassereimer86, 2022-10-11:
With the latest sources and the previous named special configuration, it's working this time! I got the error once after the update and loading the config file, but after that it didn't happen again. 🤷 I saw that I didn't replace the Version.h file last time. But I don't know if this could be the reason why I had no problems last time.

mriscoc, 2022-10-12:
Good to know that it is fixed now.

closed 2022-10-12 by mriscoc