mriscoc/Ender3V2S1 #439: Octoprint, Bed Visualizer, new mesh GCode G28 parameters issue

ebada, 2022-10-11:
**Describe the bug**
Octoprint, Bed Visualizer, new mesh GCode G28 parameters:

Something caught my eye and just want to bring it to your attention.

Invalid parameter:
**G28 XYO   ; Do an optional XY home**

M155 S30  ; set temperature reporting delay, use a value longer than the time it takes for your leveling command to complete.
G28 XYO   ; Do an optional XY home
G28 Z     ; Do a Z home
@BEDLEVELVISUALIZER  ;
.....

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#to-take-a-new-mesh-in-non-ubl-versions-use-this-code-in-the-configuration-page-of-the-plugin
3. New mesh for non ABL gcode section, line 2 
5. G28 XYO, I think this is invalid option.

**Expected behavior**
Replace line 2 and 3 with:  G28 XYZ

**Additional context**
line 2:  G28 XYO   ; Do an optional XY home
Optional XY home, where the option is present?     
 

Thank you

mriscoc, 2022-10-11:
https://marlinfw.org/docs/gcode/G028.html
[O]  | Flag to skip homing if the position is already trusted
-- | --

closed 2022-10-11 by mriscoc

ebada, 2022-10-11:
Thank you! very helpful.
I had 0 as in zero, not O :) when the editor flagged the line.