mriscoc/Ender3V2S1 #445: Intermittent characters on the display when jogging motors with SKR Mini E3 V3 board

jetcat-bob, 2022-10-14:
I first installed Marlin Professional on my Ender 3 V2 with the Creality 4.2.2 board and everything works perfectly. When I changed to the SKR Mini E3_V3 board, I got an intermittent problem with the display. When moving any of the motors, I get random characters appearing in different locations on the screen. This problem is very infrequent but annoying. For example, if I jog the Z axis from top to bottom maybe one or two times it will display a random character on the screen. However, by changing other parameters, like temperature or fan speed, the display works fine. I figured it was a hardware problem since the 4.2.2 board works fine so I exchanged the skr board for another. Unfortunately, the new board exhibits the same problem.
So now I know it is not a particular problem with the first skr board and it might be a firmware problem. I connected my oscilloscope to the display's serial port and it was exactly 115.2 kbaud. The signal looks clean at 3.3v peak to peak. The enable, A3 and 5V pins seem OK as well. Maybe there is a noise problem with the steppers / UART or a firmware issue? BTW, I tried disconnecting the stepper motors and then jogging the axis and the same problem occurred without a motor running.

Any help is appreciated!

[image]
[image]

mriscoc, 2022-10-15:
I don't have a SKR board to verify, but your test shows that could be a problem in the serial communication implementation and ISR routines for motor movement in that particular board. Sadly, there is very little I can do because that needs to be fixed at Marlin core level. Could be interesting to try with vanilla Marlin.

jetcat-bob, 2022-10-15:
Actually, tried a compile of Marlin 2.1 this afternoon. Of course the
display was not correct. Then I noticed I have the DACAI display and not a
DWIN version. So gave up not knowing if the display was compatible or not
and reinstalled my compile of your skr firmware. Now the Z axis doesn't run
in one direction.
Decided the fan noise was a feature and reinstalled the 4.2.2 board.
Everything works perfectly.....

Really like your version and not willing to compromise for the skr board.

Thanks,
Bob Wilcox

mriscoc, 2022-11-25:
It seems that the SKR Mini E3 V3 has some problems with interrupt management which can have several adverse effects like the screen glitches and BLTouch, check: https://github.com/MarlinFirmware/Marlin/issues/24351 and https://github.com/mriscoc/Ender3V2S1/issues/509

jetcat-bob, 2022-11-27:
> and BLTouch, check: MarlinFirmware/Marlin#24351

Thanks Miguel for sharing this with me. I went back to my Creality 4.2.2
board because I feared there was a possible issue with the interrupt
routines that was corrupting the code. That is still my suspicion after
reading this. I also noted the lower baud rate suggestion for the USB/UART
with the Mini 3. Ability to service the UART interrupt fast enough????
Anyway, I have Klipper running on the board now with no problems so far. Of
course it doesn't need to do much compared to running Marlin.

Thank you for your help,
Bob Wilcox

eldeeb91, 2022-12-31:
I have a similar problem with creality 4.2.2. I updated the board and the screen to the latest firmware. and sometimes it also appears on the top.
[image]

jetcat-bob, 2023-01-01:
I never found a solution using Marlin sofware. I'm using my board with Klipper now. I don't use Klipper screen but I think the Raspberry Pi handles that anyhow. Sorry I can't be of help!

closed 2023-05-04 by github-actions[bot]