mriscoc/Ender3V2S1 #449: C104 U1 T### No longer working

Zyzzbarth333, 2022-10-16:
Hey, there I hope you're well,

I have recently updated my Ender 3 V2 to Ender3V2-427-BLTUBL-20221002.bin. Before the update, I used the Sprite Extruder Pro Kit, and the C104 code worked well in unlocking the temp @300C. However, I have now changed to A Dragon Hotend, and while using C104 gets accepted, the Max Temp remains at 260C. Is there something I am overlooking?

I attempted to use the T13 version but kept getting a proper failure error.

mriscoc, 2022-10-16:
Ender3V2-427-BLTUBL-20221002.bin is working correctly:
```
[...]
Send: C104
Recv: echo:; Max Extruder temperature:
Recv: echo:  C104 T275
Recv: ok P15 B15
[...]
Send: C104 U1 T300
Recv: echo:; Max Extruder temperature:
Recv: echo:  C104 T300
Recv: ok P15 B15
[...]
```

closed 2022-10-16 by mriscoc

mriscoc, 2022-10-16:
Sprite Extruder Pro has another thermistor type.  Please open a new issue with detailed information about the error that you have with the T13 version.

Zyzzbarth333, 2022-10-16:
Hi there, I did not use the T13 Mode with the Sprite Extruder Pro.

I used;

C104 U1 T365;
M500;

and loaded it into an SD Card when I previously used the Sprite Pro and for the current config., the only thing that has changed is the hardware.

mriscoc, 2022-10-16:
> Hi there, I did not use the T13 Mode with the Sprite Extruder Pro.
> 
> I used;
> 
> C104 U1 T365; M500;
> 
> and loaded it into an SD Card when I previously used the Sprite Pro and for the current config., the only thing that has changed is the hardware.

T365 could be too high.