mriscoc/Ender3V2S1 #452: #define MANUAL_FEEDRATE extruder setting not working

tpups, 2022-10-18:
**Describe the bug**
Changing the E0 value in #define MANUAL_FEEDRATE has no effect on the feedrate when extruding using the LCD screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Prepare'
2. Click on 'Move Axis'
3. Move the extruder
4. Observe speed

**Expected behavior**
Extrude rate matches setting in #define MANUAL_FEEDRATE.

**Version (please complete the following information):**
2.1.0.1 MRiscoC Ender3V2-422-MM, based on bugfix-2.1.x

**Additional context**
Not sure if this is a user error, a Marlin bug, or a bug specific to this firmware. When extruding using the LCD menu, the feedrate seems to be about 1mm/s no matter what I set the value to in MANUAL_FEEDRATE. Do I need to reflash the LCD firmware when changing this option?

mriscoc, 2022-10-18:
Any changes to the source code, configuration.h and configuration_adv.h files require recompiling the firmware and flashing the binary.
Currently `MANUAL_FEEDRATE` only has an effect when you compile in MarlinUI mode.

closed 2022-10-18 by mriscoc

tpups, 2022-10-18:
I have been compiling, yes. I wasn't aware it was only used in MarlinUI mode, thank you for the clarification!