mriscoc/Ender3V2S1 #473: Tune/"Firmware Retract"

GT-Geo, 2022-10-30:
Under the _Tune_ menu there is "Firmware Retract". Should that not read _Filament Retract_?

classicrocker883, 2022-10-31:
no, its meant to go to the firmware retract menu. which is retraction settings based on the firmware.
youre right that it is filament retract, however this menu is based on the firmware.

Nbrooks639, 2022-10-31:
you also have the option to select firmware retraction in your slicer.
[image]

mriscoc, 2022-10-31:
It is always good to read the Wiki documentation: https://github.com/mriscoc/Ender3V2S1/wiki/Firmware-Retraction

closed 2022-10-31 by mriscoc

GT-Geo, 2022-10-31:
Sorry about that........apparently I have reading/learning to do.