mriscoc/Ender3V2S1 #478: Status Bar

pacocatech, 2022-10-31:
[image]

Hello, I downloaded the newest version of your firmware (source code), I made some changes in the speed issue and I did the build normally, however, when I install the firmware on the machine, the status bar with the shortcuts I choose does not appears on the display.
The display is also up to date with the latest version you have made available.
Is there any secret to doing the build with the status bar part?

I'm using an Ender 3 S1 - F1 and I also tested it on an Ender 3 V2 4.2.2 board

DanMan32, 2022-10-31:
I believe you need to install the "Giadej compilation" display asset rather than "stock" for the shortcuts to appear.
Also since you have an S1, you probably have the DACAI display rather than the DWIN.  You need the 'private' folder in your SD card rather than the DWIN-SET.

BakeJake, 2022-11-01:
I'm having this same issue, and oddly I haven't changed anything but the name of the printer, since the last compile anyways.
I do have it connected to GitHub though and Idk what updated but when I compile on my own ProUI doesn't pop up on reboot and the toolbar icons don't show. I can however select them still! Really weird and glad to see its not just me! Ive had no issues before and my screen is updated as well. Grabbing the bin from his releases works though.... I like to tune in my changes in the firmware rather than in the eeprom!

classicrocker883, 2022-11-01:
you must leave the author name to this `STRING_CONFIG_H_AUTHOR "Miguel A. Risco-Castillo (MRiscoC)" `

also dont forget to leave `#define ProUIex 1` and `Has_Toolbar` uncommented

pacocatech, 2022-11-01:
Thanks @classicrocker883

closed 2022-11-01 by pacocatech

BakeJake, 2022-11-01:
Cheers, thanks for that!