mriscoc/Ender3V2S1 #480: Intermittent boot loop - delay with SD Card

DanMan32, 2022-11-01:
When booting up Ender 3v2 with 8GB SD Card that came with the printer, intermittently may go into a boot loop where it will reboot a few times.

It doesn't seem to matter if the card has nothing on it or has gcode files.   More pronounced problem if rebooting printer from menu rather than powercycle.

Sometimes boot loop would only occur a couple times, sometimes seeming endless.

8GB Card formatted with Windows 10, FAT32, 4096 byte cluster size.
Doesn't matter if card is quick format or full formate

Boots OK if no card is inserted, usually reads card OK if inserted after fully booted and mesh autoloaded as sign it has completely booted up.
At one point though if I booted the printer, waited for "mesh 0 loaded", inserted card with cursor on Media icon, then pressed to select media, it would reboot.  In this scenario, would not reboot if I first moved cursor to Prepare, inserted card, then went back to media.

By reboot, evidence being that firmware details popup would appear.

I wish I could give step by step example, but situation never seems to be consistent.

Related question to SD Card, what is menu selection "Media Update" in advanced menu?  Above problem happens whether this is checked or not.

mriscoc, 2022-11-01:
**Version (please complete the following information):**
Firmware version and release date, **link to the downloaded binary file**.

**Additional context**
Add any other context about the problem here.

DanMan32, 2022-11-01:
Binary used is Ender3V2-422-BLTUBL-20221002.bin
Screen binary, if that has any influence on this, is Giadej compilation, private folder since apparently my EE3v2 apparently has the DACAI display.
It didn't start right away, but once it started, can't get it to stop.  I suppose it could be a slow SD card, got more coming tomorrow.   Even if the stock 8GB card is on the slow side, shouldn't happen since Windows produces no errors at all with the card.

I did test to see what Media Update was, perhaps that has something to do with it?  However enabled or disabled, once it started happening, it was definitely repeatable.

What does Media Update do exactly?

mriscoc, 2022-11-01:
If you don't have a card extender do not use `Media Update` feature, it resets the whole file system of the card each time that you enter in media menu and it is incompatible with some uSD cards.

DanMan32, 2022-11-02:
Thanks for indicating what Media Update is for.
But I had turned it off as well as on and still had boot problems with SD card inserted.

Except today I cannot reproduce it.  Understanding UBL and some of the other related options better, I did fine tune tramming and UBL mesh, so perhaps I had a weird minor corruption in EEPROM that finally cleared out.

mgadei, 2022-11-08:
Hi, me too, I had the same problem with [Ender3V2-422-BLTUBL-T5-LA-MPC-20221002.bin].
Booting few times unable to get out from the boot loop without a hard reset. Of course no bin file was on the sd card.  Returning to [Experimental-Ender3V2-422-BLTUBL-T13-20221002.bin] and this problem dissapeared.  Hope it helps spotting where is the cause. I have an ender 3 v2 with a Sprite extruder and cr touch and wanted to give a try to that version with Linear advance. It worked, not great, some rounded corners and detail lost, but having this issue (rebooting randomly) I did not wanted to play more  with the parameters that  I already did (flow calibration again, K, etc ).  Impressive job anyway, maybe it was my machine.

DanMan32, 2022-11-08:
In my case though it wasn't with the LA firmware but rather firmware for BL-UBL.
It would boot normally if I didn't have a card installed at all on power up or removed the card during the loop.
It stopped doing this the next day, but I was fine-tuning UBL where in the process saved config changes so maybe it was bad data in the EEPROM.

ariatpi, 2022-11-30:
I've seen hangups during boot on the stock firmware of the Ender 3 V2 Neo, so this may be unrelated to anything added in the Professional firmware.

mriscoc, 2022-12-05:
No more user feedback

mriscoc, 2023-04-08:
Informative link: https://www.3dprintgorilla.com/ender-3-pro-v2-neo-s1-sd-card-size