mriscoc/Ender3V2S1 #485: UBL Probing error

TheCableGuy99, 2022-11-02:
**Describe the bug**
I am using UBL 9x9 and when probing the bed with "G29 P1" it always fails on just 1 attempt. I have confirmed this on 3 different machines, an Ender-3v2 and 2 different Ender-3 S1 Plus's. What happens is that the probe retracts, it fails and lowers onto the bed and then re-probes the same place successfully. 

Sometimes it retracts when moving to the next position like in the attached video and other times it probes then retracts (set to probe each point 3 times). 

It is consistent in that it fails at the exact same probe number time each time but not the same position. For example...

Ender-3v2 - Fails on position 65/81, but the position on the bed might be different for this probe number as it moves around randomly.
Ender-3 S1 Plus - Fail on position 63/81, but the position on the bed might be different for this probe number as it moves around randomly.
Second Ender-3 S1 Plus - Also fails on position 63/81, but the position on the bed might be different for this probe number as it moves around randomly.

As the probe number is the same on each Ender-3 S1 Plus but different to that of the Ender-3v2 I wonder if it's something to do with the bed size.

**To Reproduce**
Heat bed to 60°C and wait for it to reach temp
Run "G29 P1" and monitor to check all probe points were successul

**Expected behavior**
To be able to build the mesh without errors

**Screenshots**
Video attached
https://user-images.githubusercontent.com/39552752/199599438-a230e72c-ad15-44e0-928d-6e903116ce77.mp4

**Version (please complete the following information):**
Version 2.1.3 20221019 - 01:40

**Additional context**
I filed a false positive yesterday which turned out to be a faulty CR-Touch but I have had that replaced today and it's working fine now. I don't think we can blame this on a faulty CR-Touch as it's happening on 3 machines.

mriscoc, 2022-11-03:
https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch

TheCableGuy99, 2022-11-03:
So I'm just in the middle of testing this on all printers...

I disabled HS mode and it works fine on all 3 printers but it's painfully slow, so I'm moving on to trying to increase the feedrate to 960 mm/min as recommended.

My config shows:
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 20, 45 }  // Ender Configs  // Ender Configs

I presume the "20" is mm/s and therefore 1200 mm/min which is already more than the suggested 960 mm/min. If so, should I try reducing it to 960 mm/min or is that pointless?

MarkforU, 2022-11-03:
I had same problem on Ender 3 V2 with 4.2.2 and another Ender 3 converted into a V2 with the 4.2.7 board. I still use HS Probing, but I do a 8x8 and don't have that failure anymore. I don't know if it had anything to do with the 9x9 original problem by Marlin 2, but when I did 9x9 3 probes I would get a crazy raise on the far right side when I had to print something large on the bed. I also found that 3 probe touch was needless since I tested probe accuracy when doing a probe test, so I set it to probe twice which was I believe 1 on my the Ender. Anyway since new firmware release, I haven't tried 9x9 and still use 8x8 with HS, Probe touches twice and I get no errors. Hope that helps in figuring this out.

TheCableGuy99, 2022-11-03:
> I had same problem on Ender 3 V2 with 4.2.2 and another Ender 3 converted into a V2 with the 4.2.7 board. I still use HS Probing, but I do a 8x8 and don't have that failure anymore. I don't know if [...]

Thanks for all that info. I managed to get it working with 7x7 on all 3 printers on HS mode, it wouldn't work with 8x8. I would prefer mode probe points for better accuracy so wonder if I switch to 2 probes instead of 3 it might work with 8x8 like yourself as well? Or even 9x9. I'll give it a go. Thanks again.

TheCableGuy99, 2022-11-03:
I can't see a custom gcode for the number of probe times ( https://github.com/mriscoc/Ender3V2S1/wiki/Custom-C-gcodes ), is there a marlin code for it?

MarkforU, 2022-11-03:
Under the probe menu on how many times I want it to touch. Also at the bottom of the probe menu there's a Command to run a M39 or M42...can't remember, Probe Test. It will tell you the accuracy of your probe. I'm running the latest CR-TOUCH that was released just back at the beginning of the year. Compared to a old CR-TOUCH that I had from a year and half ago, I ended up upgrading my older one. My probe touches each spot twice in a 8x8. I have touched the z-axis speed since new firmware, so I believe it was reset to default beings my firmware update set everything else back to default numbers...so I have listing for probe touches to 1 which gives me 2 touch spots. Although I fired my Ender 3 with 4.2.7 board on it, I noticed it was set to 1, which is where I want it. When I had it build a mesh, it only probed 1 touch...so something was strange about it, so I might need to change it to a 2 for probe on that machine. I do notice some differences in menu and responses between the 4.2.2 and 4.2.7 boards. If I replace this board again, I'll switch it to a 4.2.2 board. For instance when tramming on the 4.2.7 board, it wants to hit more to the outer corners of the bed. On 4.2.2 board, it trams right on top of where the screw would be to adjust the corners. Both machines are almost identical in numbers...only numbers or perimeters that are different between my 2 machines is my extruder numbers.

TheCableGuy99, 2022-11-03:
So I found "multiple probing" in the menu and change it from 2 to 1. But then it jumped from 3 to 1 probes. The good news is that it did a 9x9 (81 probes) successfully. I'd be willing to bet if I can find out how to do 2 probes it would also do an 8x8 sucessully.

I looked into the the config file for multiple probing and found this:
/**
 * Multiple Probing
 *
 * You may get improved results by probing 2 or more times.
 * With EXTRA_PROBING the more atypical reading(s) will be disregarded.
 *
 * A total of 2 does fast/slow probes with a weighted average.
 * A total of 3 or more adds more slow probes, taking the average.
 */
#define MULTIPLE_PROBING 2  // MRiscoC Enabled
#define EXTRA_PROBING    1  // MRiscoC Enabled

I'm just not sure how to alter this to make it probe twice.... maybe @mriscoc can clarify what to set it to?

MarkforU, 2022-11-03:
Excellent, you're getting somewhere. Go into your probe settings menu. Scroll to very bottom, it gives you a M48 probe test. My results after 10 test probes it runs was 0.003202 of a deviation. That's isn't so bad, I wasn't worried about doing 3 touches. I just checked my 4.2.7 board that I just tested. As o said I use to leave it on 1 and I got 2 probe touches. I said I was only getting 1 touch from the probe, and just now realized the firmware did set it back to 0 for multiple probes. So I just changed it to 1 now, I'm getting 3 total touches, which should only be 2. Just got done with a print on the Ender V2 and also saw it set to 0...never really payed attention to its settings after I set it up. What I've found is if I set my 4.2.2 board to 1 probe setting and then save settings then go back to probe settings, it's back to 0 giving me only 1 probe, so, this time I just set it to 2 probes and did save, go back into menu and still read 2 and not back to 0 again. So I told it to make a mesh, now I'm getting 3 touches like it is supposed to. 
So Problem...0 saves and 2 saves but 1 multi probe that gives you 2 probe points goes back to 0 after storing it in firmware. Myself, I don't really have a problem with a single probe, I would like 2 back...but 3 probe touches also works. Never had this issue with last UBL Firmware before I just updated the other day. Just now tested the 4.2.7 board and there's no getting to 1 on it either and storing, it drops back to 0 multi probe, but when setting to 2, it saves the 2, it won't save 1 which = 2 touch points. 
Actually glad you saw this, because running UBL, I don't have to probe my bed every few Days of printing to check level, but since this issue came up...I can either set to 0 and get 1 touch, or set to 2 and get 3 touch...so you're correct the EEPROM is not saving the setting for 1 which gives you 2 touches in each spot...still on HS haven't had to slow the Z yet, still at firmware speed...which old  firmware I DID have to slow the Z speed down, I haven't had any probing errors as of yet since new firmware, but you found a bug with it not storing the 1 multi probe in the probing menu, I'm sure if you set yours to 1 and save settings, go back into Probe Menu you'll see it has been set back to 0, but entering a 2 it keeps that setting after saving and I'm still doing 8x8 grid because it's plenty, but haven't had failure to have to restart printer on the error that stops everything. 
So, since I run a bed that's locked in place without springs, I'm going to just go ahead and use 2, let it do the 3 probes on a 8x8 and wait for the next fix. UBL keeps me going until I have to change nozzles or build plate.

TheCableGuy99, 2022-11-04:
Okay I'm getting nowhere with this...

If I set it to the following, it won't compile saying "MULTIPLE_PROBING must be 2 or more":
#define MULTIPLE_PROBING 1  // MRiscoC Enabled
#define EXTRA_PROBING    1  // MRiscoC Enabled

If I set it to the following, it compiles but still probes each position 3 times with the default of "Multiple Probing" set to "2" in the menu. If I set it to "1" in the menu it jumps from probing 3 times to once:
#define MULTIPLE_PROBING 2  // MRiscoC Enabled
//#define EXTRA_PROBING    1  // MRiscoC Enabled

If I set it to the following and disable both in the firmware and set "Multiple Probing" to "0" in the menu it probes once. You can't set it to "1" under this firmware setting because when you leave that menu and go back in it changes back to "0". If I set it to "2" it jumps back to probing each position 3 times.
//#define MULTIPLE_PROBING 2  // MRiscoC Enabled
//#define EXTRA_PROBING    1  // MRiscoC Enabled

I can't work out for the life of me how you got it to probe twice at each point.

Unless I can get this to work I will have to set it to 7x7 on all printers which I'm not sure is any better than ABL, unless the algorithm is better? Or go back to ABL.

@mriscoc said he is going to stop compiling firmware for ABL but i'm not sure if that means we will always be able to compile ABL ourselves with the configurator, or will this option be removed in the future? Some clarification on this and if we can get it to probe twice per position would be nice, please?

mriscoc, 2022-11-04:
> I can't see a custom gcode for the number of probe times ( https://github.com/mriscoc/Ender3V2S1/wiki/Custom-C-gcodes ), is there a marlin code for it?

C851 Sn [Zn] Mn : Set probe Z feed speed (S) in cm/s, Multiple probing (Mn>1) or disables it (M0), in manual mesh version, Z parameter can be used to set a manual Z-offset

TheCableGuy99, 2022-11-04:
> C851 Sn [Zn] Mn : set probe Z feed speed (S) in cm/s, Multiple probing (Mn>1) or disables it (M0), in manual mesh version, Z parameter can be used to set a manual Z-offset

I'm sorry but i'm completely confused by this. The above appears to give me a command to set the Z rate in cm/s but the article you linked me to originally suggests increasing it to mm/min but i'm already past that at 1200 mm/min so I don't see how this would help?

I would prefer to do a 9x9 double probe rather than 7x7 triple probe which i've currently had to set it to.

If the above (C851 Sn [Zn] Mn : set probe Z feed speed (S) in cm/s, Multiple probing (Mn>1) or disables it (M0), in manual mesh version, Z parameter can be used to set a manual Z-offset) tells me how to set a double probe I'm not getting it, could you elaborate a bit please?

mriscoc, 2022-11-04:
> I'm just not sure how to alter this to make it probe twice.... maybe @mriscoc can clarify what to set it to?

You can't EXTRA_PROBING is currently hardcoded to 1.

>@mriscoc said he is going to stop compiling firmware for ABL but i'm not sure if that means

Future improvements will be made only of UBL code, and the ABL code will remain as it is now.

TheCableGuy99, 2022-11-04:
> > I'm just not sure how to alter this to make it probe twice.... maybe @mriscoc can clarify what to set it to?
> 
> You can't EXTRA_PROBING is currently hardcoded to 1.
> 
> > @mriscoc said he is going to stop compiling firmware for ABL but i'm not sure if that means
> 
> Future improvements will be made only of UBL code, and the ABL code will remain as it is now.
> 
> Note that for a 230x230 bed a 5x5 mesh is enough.

Okay so if a 5x5 is enough for the Ender 3v2, would a 7x7 be enough for a 300x300 bed? I was just presuming that I should be using 10x10 (or as close to it as I could get) because that's the default for UBL from the documentation I've read.

mriscoc, 2022-11-04:
>Okay so if a 5x5 is enough for the Ender 3v2

Assume a separation between mesh points of about 50mm.

TheCableGuy99, 2022-11-04:
Okay just to be sure I understand what you are saying.... Are you saying that I should try to keep it to as minimum as possible because moving the Z up and down a lot decreases it's accuracy? Also not to go more than 50mm apart in probes? If so then I should be using 5x5 for a 300x300 bed then... right?

mriscoc, 2022-11-04:
Each machine is unique, so, test both (5x5 and 7x7) if you got the same results, stay in 5x5.

TheCableGuy99, 2022-11-04:
okay, thank you for the help, i'll close this ticket now.

closed 2022-11-04 by TheCableGuy99