mriscoc/Ender3V2S1 #490: Compiling error to use BLTouch

budm, 2022-11-05:
**Describe the bug**
Im building an Ender 3 Max using v2 components including a CR-Touch/BLTouch, I've been slowly adjusting the Configuration to enable the features needed. I'm using a Mac, VSCode, and Auto Build Marlin

**To Reproduce**
Steps to reproduce the behavior:
1. In Configuration.h I uncommented #define AUTO_BED_LEVELING_BILINEAR
2. In Configuration.h uncomment #define BLTOUCH
3. In Configuration.h uncomment #define Z_SAFE_HOMING
4. While compiling in VSCode I got the following error:5. 

"Compiling .pio/build/STM32F103RE_creality/src/src/lcd/e3v2/proui/dwin_lcd.cpp.o
Compiling .pio/build/STM32F103RE_creality/src/src/lcd/e3v2/proui/dwin_popup.cpp.o
Compiling .pio/build/STM32F103RE_creality/src/src/lcd/e3v2/proui/dwinui.cpp.o
Marlin/src/lcd/e3v2/proui/dwin.cpp: In function 'void SetZOffset()':
Marlin/src/lcd/e3v2/proui/dwin.cpp:2010:7: error: 'babystep' was not declared in this scope
 2010 |       babystep.accum = round(planner.settings.axis_steps_per_mm[Z_AXIS] * BABY_Z_VAR);
      |       ^~~~~~~~
*** [.pio/build/STM32F103RE_creality/src/src/lcd/e3v2/proui/dwin.cpp.o] Error 1
====== [FAILED] Took 282.45 seconds ======

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:04:42.455
= 1 failed, 0 succeeded in 00:04:42.455 =
md@M-MacBook-Pro Ender3V2S1-20221002 % 
"

**Version (please complete the following information):**
Ender3 V2/S1 Professional Firmware 20221002

**Additional context**
Copy of my Config
[Configuration.h](https://github.com/mriscoc/Ender3V2S1/files/9944361/Configuration.copy.txt)

classicrocker883, 2022-11-07:
is `BABYSTEP_ZPROBE_OFFSET` defined in Configuration_adv.h?

budm, 2022-11-07:
goodness, that was it! Appreciate it!

closed 2022-11-07 by budm