mriscoc/Ender3V2S1 #494: G-code Preview Setting in source code

CJ65CM, 2022-11-09:
Is there something in the source code that needs to be "turned on" that I am unaware to get the G-code preview option to work?

After having used this Firmware for a few months now liking it, I picked up an Ender 5 Pro for next to nothing at an estate sale this past weekend and decided to try to adapt the Firmware using the source code provided for the 4.2.2 board since that is the board that was in the Printer and change it to using the DWIN type display instead of the regular Ender 5 LCD type display and I got everything working as intended BUT the G-code preview doesn't work on this printer.

I've tried swapping displays between printers (the G-code preview works as it should on my Ender 3V2) and the preview image shows up fine on both screens on the Ender 3 but on the 5 that I adapted the source code, both screens only show the QR Code Image which leads me to suspect that there is a line of code that I've overlooked that needs to be changed or uncommented or something to get the preview to work on the 5 like it does the 3.

Cant anyone turn me in the right direction here with what I may have missed that needs to be corrected in the code?

mriscoc, 2022-11-10:
Did you change the firmware author? ProUI extension library needs to have some unchanged values.
Can you see the icons in the Shortcut tool bar?

mriscoc, 2022-12-05:
No more user feedback

closed 2022-12-05 by mriscoc

CJ65CM, 2022-12-06:
Hi, I'm sorry for this terribly late response. I thought that I had came back on and replied after I had read your initial comment and investigated the problem more but apparently I didn't.

Anyways, yes that was the problem (Firmware Author). 
My Son (he's 11 and trying to learn coding) initially had started making the changes that we were going to need to make it work with the Ender 5 and got confused along the way so he handed it over to me and it turned out that he had changed the Firmware Author to himself along with changing the display name.

Again, sorry for the late response to this and thank you for the input.