mriscoc/Ender3V2S1 #499: the firmware don't recognize my sd gcode

scyther149, 2022-11-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. i saved the .gcode with the script correctly
2. i turn on the ender 3 v2 and the machine don't recognize my .gcode

**Expected behavior**
see my gcode with a thumbnail in the print screen.

**Screenshots**
the script installed correctly

[image]

all the .gcode in the sd

[image]

the print screen

[image]

the firmware version

[image]

and i have other 3v2 with another firmware, and that firmware can read the gcodes fine, i only have this error with this firmware (i was trying appears the thumbnail feature)

[image]

**Version (please complete the following information):**
"Ender3V2-422-MM-20221002.bin"

**Additional context**
I have Cura 5.0.0.

classicrocker883, 2022-11-10:
if the firmware works on other machines, then this is only particularly related to that machine. either the SD card isn't formatted correctly, or your SD card reader is malfunctioning, like a solder joint may need to be reflowed.

scyther149, 2022-11-11:
> if the firmware works on other machines, then this is only particularly related to that machine. either the SD card isn't formatted correctly, or your SD card reader is malfunctioning, like a solder joint may need to be reflowed.

the sd is formatted fine because it work in other my other machines... and the sd card read it fine when i installed other firmware, only don't work with this in particulary, if i come back to fabric or i install other firmware it works again...
there's something like a bug with a specific series i think, the machine that don't work is my newest ender 3 v2, but is weird because is 4.2.2 too, like the others 2 olders

MarkforU, 2022-11-11:
I got newest V2 and installed with a SanDisk 32gig card, fast card, newish card. Also have a Ender 3 modded has, 4.2.7 board in it. No problems. The thing is, that slot you drag your card in and out of, my first machine the board wasn't in there correctly and the card went right over the top. I'm guessing also, a bad solder joint on its not making correct contact, or a bad card...don't use Creality 8gig cards either, they fail at any given time.

mriscoc, 2022-11-11:
Do your printers have any hardware differences?

scyther149, 2022-11-11:
> Do your printers have any hardware differences?

the 3 screen motherboard looks pretty similar, and the 3 printers are 4.2.2, is it okay? or what do you mean with a hardware difference?

[image]

PD: Thank you for respond me, Miguel! I'm from Argentina, we can continue in spanish if you want... I sent you a message in facebook a few days ago, i wanted personalize the visual appart of the printers, i did it with other firmware that you made a post i think "E3V2-Default-v4.2.2-v2.0.1" + DWIN_SET(Original), when i upgraded this "thumbnail firmware" (Ender3V2-422-MM-20220816) almost all the icons work, but some of they don't, they crashed, that's because the name of the jpg inside the 9.ico don't are the same i think, can do you share with me the 9.ico of this firmware? the dwin_set folder i mean... i'm doing some skins of pokemon for the printers hahaha

[image]

mriscoc, 2022-11-17:
>what do you mean with a hardware difference?

I am referring, for example, to memory card extenders. There is a item in Advanced settings menu: `Media Update` that force to re-read the file system of the memory card when listing files.

About screen icon assets see here: https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released/display%20assets

mriscoc, 2022-12-05:
No more user feedback

closed 2022-12-05 by mriscoc