mriscoc/Ender3V2S1 #500: Y axis randomly stops working?

classicrocker883, 2022-11-10:
this one is difficult to reproduce. it can happen during printing. ive made it happen when in the Mesh Edit menu, select probe Z reference value, and the Y stepper motor freezes at random. so it may not always stop working after these steps, but disabling motors resets it and it works normal again.

It may have something to do with the planner buffer, or synchronize, or parser, or one of the gcode or probe files.

I'll see if I can find more about it, maybe it has happened to someone else.

classicrocker883, 2022-11-11:
>Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
Recv: X:142.50 Y:83.67 Z:11.75 E:0.00 Count X:11397 Y:6697 Z:4700

when it says
`Recv: Bed X: 10.00 Y: 210.00 Z: 0.04` the screen status shows `X: 10.00, Y: 210.00, Z: 0.04`

and what does this mean?
`Count X:4399 Y:17369 Z:4700`

classicrocker883, 2022-11-12:
the BL touch error / Probing failed happens with `ProUIex` is defined. so this doesnt happen when the firmware is without the proui extra features.

classicrocker883, 2022-11-13:
@mriscoc take a look at file src/module/probe.cpp

line ~740
`  #elif TERN(ProUIex, 1, Z_PROBE_FEEDRATE_FAST != Z_PROBE_FEEDRATE_SLOW)`

this is under `#if !ProUIex` so the above TERN becomes redundant whether or not ProUIex is ENABLED or not.

i think this stems to the problem because it works fine without ProUIex enabled, so can `run_z_probe` be enabled since it is needed for the `return measured_z` function? this is the reason why there is Probing Error, because this error pops up
line ~910
>if (isnan(measured_z)) {
    stow();
    LCD_MESSAGE(MSG_LCD_PROBING_FAILED);
    #if DISABLED(G29_RETRY_AND_RECOVER)
      SERIAL_ERROR_MSG(STR_ERR_PROBING_FAILED);
    #endif

mriscoc, 2022-11-17:
>line ~740
#elif TERN(ProUIex, 1, Z_PROBE_FEEDRATE_FAST != Z_PROBE_FEEDRATE_SLOW)
  
Yes, that is redundant, that code comes from when was not posible to edit Multiple Probbing, change it to:
```cpp
#elif Z_PROBE_FEEDRATE_FAST != Z_PROBE_FEEDRATE_SLOW
```

Changing the source code could sometimes break the internal structure of the variables and constants of the ProUI extensions library. For that cases is better to define ProUIex to 0 in Configuration.h

https://github.com/mriscoc/Ender3V2S1/blob/1c733afd3d533959f6822341fadfc77dc88a9e72/Marlin/Configuration.h#L3160

closed 2022-11-17 by mriscoc