mriscoc/Ender3V2S1 #502: Power Loss Recovery option missing from Ender 3 S1 UBL build

TUGTEN, 2022-11-12:
**Describe the bug**
The power loss recovery option is missing from the latest Ender 3 S1 UBL build. It is however, present on the non-UBL build.

mriscoc, 2022-11-17:
Some features were disabled to make space for UBL in S1 F4:
BAUD_RATE_GCODE
INDIVIDUAL_AXIS_HOMING_SUBMENU
SOUND_MENU_ITEM
LONG_FILENAME_WRITE_SUPPORT
BINARY_FILE_TRANSFER
**POWER_LOSS_RECOVERY**

closed 2022-11-17 by mriscoc

TUGTEN, 2022-11-21:
Power Loss Recovery does not work even after enabling it in the settings. (Non-UBL version) 

Just to try it out, I started printing a calibration cube, and powered off the Ender 3 S1 after the 5th layer. On the stock firmware, after power was switched back on, the prompt to resume print appeared. However, after repeating the same steps on the Professional Firmware, the prompt did not appear.

mriscoc, 2022-11-22:
> Power Loss Recovery does not work even after enabling it in the settings. (Non-UBL version)
>
> Just to try it out, I started printing a calibration cube, and powered off the Ender 3 S1 after the 5th [...]

After turn off the printer, remove the SD card and check in a PC if there is a file named "PLR", that is the recovery file, if that file is not written or corrupted the printer will no resume the print job.

sdlafferty, 2023-03-22:
Is there much chance in future that some, or all, of these options could be reintroduced into the UBL F4 firmware? Any other features that are likely to have their memory space requirements cut down?

mriscoc, 2023-03-22:
Only in custom compilation is possible to fine tuning what features to add, but F4 doesn't allow more than one big feature.