mriscoc/Ender3V2S1 #504: Ender 3 S1 Pro, dead stock, which version?

lamachine, 2022-11-13:
I have decided the Creality firmware is crap.  I used Marlin on previous printers and want to use it on this.

Updating the firmware with Creality requires a special version for the pro, and required an update to screen software.  

Does this software work, and if so, what version?  When I search, I cannot find reference to the pro version.  

Just going with the S1 F4 for now.  Fingers cross and thanks for building all of these.

lamachine, 2022-11-13:
Well, that put me one step closer to bricking my printer.  

There appears to be a thing unique to the Ender 3 S1 Pro that requires the bin file be in a folder called STM32F4_UPDATE.  In the Creality instructions, you need the file at root level along with that folder, and the same file again in the folder.  I loaded it and it basically painted the points of light that spells out Creality, but never draws the line and never finishes booting.  

Now going to try moving back to OEM firmware, but hoping you may have some insight for me.  Your's is THE go-to firmware for these machines.

mriscoc, 2022-11-13:
This firmware is not compatible with touch screens.
Relevant info: 
https://github.com/mriscoc/Ender3V2S1/wiki#requirements
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware

Chihhao, 2022-11-14:
I also want to know which version of mainboard firmware i should use at Ender 3 S1 Pro.
When using the new version firmware, is it still possible to use the touch screen (with stock screen firmware)?

lamachine, 2022-11-14:
@mriscoc Thanks.  The Pro version seems to throw everything off.  Replacing the Marlin with the OEM resolved the issues I started with.  Can I use the Marlin if I use OctoPrint instead?  I know I probably won't want to for reasons like checking the status and setting up the machine locally, but just wondered.  

@Chihhao For me, original firmware, original screen works.  Updated OEM firmware update screen didn't work first time, but works now.  Marlin firmware with updated screen didn't work.

lamachine, 2022-11-14:
@Chihhao Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.24F4_FDM_LASER was the version that is running now.

closed 2022-11-14 by lamachine

Chihhao, 2022-11-14:
so you used the "Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.24F4_FDM_LASER" ?
does it support otcoprint perfect?

lamachine, 2022-11-14:
I have no idea, I have not connected it to that yet.

Chihhao, 2022-11-14:
i found this but haven't try it.
[https://www.th3dstudio.com/product/creality-ender-3-s1-12864-lcd-conversion-upgrade-kit/](url)
but it ditched touch screen, i feel a little stupid.

mriscoc, 2022-11-14:
You can use the color screen with the knob encoder of the S1 or V2