mriscoc/Ender3V2S1 #507: Unable to cancel print on printer when filament runout sensor trigger

lin-ycv, 2022-11-18:
**Describe the bug**
When filament runout sensor is triggered, there's no way to cancel the print, the only option is to load filament, but sometimes there's other problems that needs to be solved, so it'd be nice if there was a cancel print option when the sensor is triggered.

**To Reproduce**
Steps to reproduce the behavior:
1. start a print without filament loaded
2. sensor triggered after start gcode
3. unable to cancel print from printer other than power off

**Expected behavior**
an option to disable filament sensor when triggered, or to cancel the print.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
[Ender3V2-422-BLTUBL-20221002.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20221002/Ender3V2-422-BLTUBL-20221002.bin)

**Additional context**
Add any other context about the problem here.

lorenzomoon, 2022-12-18:
Confirmed with creality filament sensor and Ender 3 v2 Neo.

mriscoc, 2022-12-19:
The latest version has a Runout sensor enable/disable menu item in the tune menu

mriscoc, 2022-12-22:
If the M600 process (change filament) is already ongoing there is no option to cancel the print in the middle of it. I will reopen this issue for future research.

RaffaeleO, 2023-01-06:
Ender 3 V2 Neo, estrusore creality Sprite Pro e firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin].
Il sensore viene rilevato senza alcun problema e viene visualizzato anche nel monitore della stampante. Il grosso problema è che nonostante venga rilevata la rottura o esaurimento filamento, la stampa non va in pausa. continua a stampare senza filamento. Il firmware sarebbe il top se si risolvesse questo Bug e il bug del ripristino stampa dopo interruzione corrente.
Grazie

RaffaeleO, 2023-01-06:
Ender 3 V2 Neo, creality Sprite Pro extruder and firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin].
The sensor is detected without any problems and is also displayed in the printer monitor. The big problem is that although the filament breakage or runout is detected, the print does not pause. continue printing without filament. Firmware would be top if this bug and print restore after power outage bug were fixed.
Thank you

mriscoc, 2023-01-08:
> Ender 3 V2 Neo, creality Sprite Pro extruder and firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin]. The sensor is detected without any problems and is also displayed in the printer monitor. The big [...]

Reduce the filament runout distance in filament settings menu.

RaffaeleO, 2023-01-16:
Thank you for your kind reply, but unfortunately even if I reduce the
distance to 1 mm unfortunately the print does not pause. I also take
this opportunity to confirm that in the event of a blackout, printing
restarts from home and not from where it left off. I can not solve
even interacting on file configuration.h
Could you give me some advice for both problems?
Thanks and sorry to bother you

mriscoc, 2023-01-23:
> but unfortunately even if I reduce the distance to 1 mm unfortunately the print does not pause.

In Advanced / Runout diagnostic you can debug your filament sensor. See about filament settings menu in the Wiki.

RaffaeleO, 2023-01-23:
Hello Miguel, meanwhile thank you for your availability, filament
sensor problem solved, I was a defect of the sensor hardware and not
of your professional firmware. What remained a big problem is the
resumption of printing after the power outage, unfortunately when
printing resumes after several nozzle cleanings it starts from home,
in fact I risked 3 times (making attempts) to break holden.
There is probably some bad code in the gcode. I also tried to check
the file with visual studio code but I can't find the dedicated
section. Thanks in advance for the help. Raffaele Ommeniello (Italy)

RaffaeleO, 2023-04-23:
With the latest firmware everything goes perfectly, no problems encounteredInviato dal mio Galaxy
-------- Messaggio originale --------Da: "github-actions[bot]" ***@***.***> Data: 23/04/23  04:24  (GMT+01:00) A: mriscoc/Ender3V2S1 ***@***.***> Cc: RaffaeleO ***@***.***>, Comment ***@***.***> Oggetto: Re: [mriscoc/Ender3V2S1] Unable to cancel print on printer when filament runout sensor trigger (Issue #507) 
This issue has had no activity in the last 60 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 10 days.

closed 2023-07-04 by github-actions[bot]