mriscoc/Ender3V2S1 #512: Temperatur 0 after Temp is in Top

gorillale, 2022-11-21:
- i set Temperatur Bed and Hotend
- after i print a model
- the temp will be up on top
- when printing starts, both temp goes to 0
- printing with 0 temp not work

I print with cura to repetier-server pro to the creality ender 3 v2 blt 4.2.2

Mriscoc Version 4.2.1 ender 3 v2 blt

mriscoc, 2022-11-22:
Check your start and end g-code, also verify if you can mimic the g-code scripts of Octoprint 
https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts
https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#octoprint-gcode-scripts

gorillale, 2022-11-26:
thank you, but i have this settings. 
Not known where make this temp reset... 
After is print/send Cura print to the repetier on ender.
After 1 min. i close cura then this error come.
When i not close Cura, ther is no error.
All code ist send before to repetier and i not known how is the problem.

I will be ask repetier for this problem.
thank you.

mriscoc, 2022-12-23:
Is there any news about this?

gorillale, 2023-04-23:
You can close please, never see this. I have updated Cura to 5.3.0 and newst Repetier.
Thank you fpr your work.
Regards

closed 2023-04-23 by gorillale