mriscoc/Ender3V2S1 #513: Creality Cloud-Octoprint Plugin in Ender3V2-422-BLT not working

wangrui1573, 2022-11-21:
When I use the firmware Ender3V2-422-BLT in Ender3 V2, the Octoprint Creality Cloud-Octoprint Plugin does not work, it repeatedly prompts that the model fails to obtain, even if I manually modify the model, can I customize the model or device name in the firmware

I think if I can change the name in the firmware to the official name, maybe the Creality Cloud-Octoprint Plugin will work

[image]

mriscoc, 2022-11-22:
I did not test the firmware with that plugin, maybe you can request to the plugin developer what the printer firmware needs to be recognized.

wangrui1573, 2022-11-23:
Thanks , I modified the plugin's script to use manually specifying the devicename instead of reading the machine value and it's back to normal now

wangrui1573, 2022-11-25:
fixed:
https://github.com/wangrui1573/OctoPrint-CrealityCloud-With-Video

closed 2022-11-25 by wangrui1573