mriscoc/Ender3V2S1 #517: Z-Ofset not working (SKRMINI E3 V3)

gsuresh2u, 2022-11-23:
Sir, i changed my Ender3 V2 board to SKRMINIE3_V3
and i am using this firmware on that new board
i have used special configurations for this board
only i have enabled PSU control for Auto Shutdown after print
my Z-Offset is 3.23 mm (using Microswiss Direct drive Hotend)
after setting that 3.23 , stored settings, and restarted printer
that 3.23 offset is showing fine
But when start print, it is starting print in mid air, without takign that offset
i am attaching my Configs
Please check
[Configs.zip](https://github.com/mriscoc/Ender3V2S1/files/10073429/Configs.zip)

mriscoc, 2022-11-23:
In a terminal try
```gcode
G0 Z0 F300
```
Z-offset with a probe like bltouch is a NEGATIVE number.

mriscoc, 2022-11-23:
So maybe the problem is not Z-offset is your Z-home.

closed 2022-11-24 by gsuresh2u