mriscoc/Ender3V2S1 #532: Progress bar offset?

LunarstarPony, 2022-11-30:
**Describe the bug**
Progress Bar doesn't seems right on Ender 3 V2 Neo?

**To Reproduce**
Steps to reproduce the behavior:
1. Just start Print

**Expected behavior**
I would hope the Progress bar being inside the square instead xD

**Screenshots**
[image]

**Version (please complete the following information):**
20221002

**Additional context**
Or is that a settings I need to change myself pretty new in 3D Printing.
Oh and as far as what I see V2 Neo not showing Model Preview should be normal?

mriscoc, 2022-11-30:
That can only be fixed If the screen is DWIN or DACAI with the icon set of: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

closed 2022-11-30 by mriscoc

LunarstarPony, 2022-11-30:
That does indeed fix it Thank you! < 3

LunarstarPony, 2022-11-30:
BTW If I understand correctly there's currently no way for DACAI display to show Model Preview?

mriscoc, 2022-11-30:
> 

We need that Creality release the source code of Ender3V2 Neo to be able to use the DACAI API

LunarstarPony, 2022-11-30:
hmmm Alright  Thx eitherway!