mriscoc/Ender3V2S1 #541: Printer Drawing the first line outside the bed with the provided G Code Script?

LunarstarPony, 2022-12-08:
**Describe the bug**
Printer Drawing the first line outside the bed with the provided G Code Script?

**To Reproduce**
1, Using the provide G Code Script for Prusa Slicer
2, Print

**Expected behavior**
Draw the line inside the bed

**Screenshots**
[image]
[image]

**Version (please complete the following information):**
Firmware version and release date
2.1.3 20221002
**Additional context**
So It does appears that it's drawing inside the bed just very close to the edge so that it's pretty much outside on the actual printer?
Ender 3 V2 Neo

classicrocker883, 2022-12-09:
you may need to adjust the X offset, but usually these printers are set to print on 220x220. did u set it to 235x235?

LunarstarPony, 2022-12-09:
You mean on the printer settings?

LunarstarPony, 2022-12-09:
If I remember correctly it's currently set to default which is 230x230

LunarstarPony, 2022-12-09:
I do have a working GCode which is get directly from Creality Slicer but I can't figure out how to get it onto Prusa Slicer
Any idea If can convert this part onto Prusa Slicer?
```
M420 S1 Z2; Use Level Mesh up to 2mm;
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

mriscoc, 2022-12-09:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

LunarstarPony, 2022-12-09:
The Max min part? I thought it's calibrated for Ender 3 V2 already

mriscoc, 2022-12-12:
Not a firmware issue.