mriscoc/Ender3V2S1 #542: Display problems

YanivKrit, 2022-12-09:
**Describe the bug**
after using trimming wizard it doesn't show the numbers on the circules and after using the build mesh feature the display is unreadable and only when I turn off the LCD the display comes back

**To Reproduce**
Steps to reproduce the behavior:
1. Go to prepare
2. Click on build mesh/ trimming wizard
4. See attached pictures 

**Expected behavior**
see the numbers after using the trimming wizard and a working display with results and buttons after using the mesh build

**Screenshots**
[image]

[image]

**Version (please complete the following information):**
Ender3S1-F4-20221002.bin

**Additional context**
Thank you for a great frimware - it makes the leveling and operation much easier

mriscoc, 2022-12-11:
Hi, we never saw that specific problem before, so, it can be a connection problem, wiring or an incompatible screen firmware. It is not an issue with the main board firmware.

seene, 2022-12-12:
Same Problem here. 
With a brand new Ender 3 S1 F4

ugodnihc, 2022-12-28:
Does this screen issue affect the actual operation of the firmware past this levelling issue? I'm unable to see the numbers, but am able to still get the tramming wizard to confirm that it is flat - are the results dependable with an incompatible screen?

seene, 2022-12-28:
So far, no problems while printig for me

YanivKrit, 2023-01-07:
a quick update - just installed fw [Ender3S1-F4-20221222.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3S1-F4-20221222.bin) and the display seems better - during calibration the display only missing the numbers but the buttons and the rest is fine but when going to "view mesh" the display remains the same

[image]

MorphMixology, 2023-03-10:
I bought an Ender 3 Pro. I wanted to use this firmware and the Micro Swiss DD setup I had on my E3V2s before. I ordered this screen from Amazon thinking it was what I needed. Apparently I got the stupid screen that doesnt work right with this? Can anyone help?

https://www.amazon.com/dp/B098WFCQYW?psc=1&ref=ppx_yo2ov_dt_b_product_details

[image]

jmealo, 2023-03-13:
I'm having issues with the mesh viewer, otherwise the screen seems to work fine. It seems the F4 variant comes with the incompatible screens :(.

mriscoc, 2023-03-21:
@migsun does you TJC screen this problem also? https://github.com/mriscoc/Ender3V2S1/issues/542#issue-1487494079

mriscoc, 2023-03-21:
Hi TJC users, I want to identify some issues with TJC displays

Did you see this update:
[image]

inside of this zip? https://file2-cdn.creality.com/file/753e7215008c87f6499da6bd4e414ea5/Ender-3S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER.zip

asyxpaul, 2023-03-21:
Got the update and installed it.  Unfortunately it still didn't fix the issue with the numbers on tramming.
[image]
[image]

mriscoc, 2023-03-21:
@asyxpaul please test with this version:

[Experimental-Ender3V2-427-BLTUBL-MPC-TJC-20230321.zip](https://github.com/mriscoc/Ender3V2S1/files/11032060/Experimental-Ender3V2-427-BLTUBL-MPC-TJC-20230321.zip)

asyxpaul, 2023-03-21:
It now has numbers but looks a little off
[image]

mriscoc, 2023-03-21:
Sorry, I made a mistake, re-download and try again please.

CalvinYohannes, 2023-03-21:
> @asyxpaul please test with this version:
> 
> [Experimental-Ender3V2-427-BLTUBL-MPC-TJC-20230321A.zip](https://github.com/mriscoc/Ender3V2S1/files/11032183/Experimental-Ender3V2-427-BLTUBL-MPC-TJC-20230321A.zip)

Does this work for S1 F4?

CalvinYohannes, 2023-03-21:
> Hi TJC users, I want to identify some issues with TJC displays
> 
> Did you see this update: [image]
> 
> inside of this zip? https://file2-cdn.creality.com/file/753e7215008c87f6499da6bd4e414ea5/Ender-3S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER.zip

I tried clicking this link is not accessible
[image]

mriscoc, 2023-03-22:
Look at the Creality Cloud website

[image]

CalvinYohannes, 2023-03-22:
> Look at the Creality Cloud website
> 
> [image]

This is TJC?

I mean, does this link work for S1? Because the link only says V2
https://github.com/mriscoc/Ender3V2S1/files/11032183/Experimental-Ender3V2-427-BLTUBL-MPC-TJC-20230321A.zip

mriscoc, 2023-03-22:
> I mean, does this link work for S1?

No, please don't download or flash anything if you don't know what to do. There is a high risk of brick your printer.

mriscoc, 2023-03-22:
Sorry. I deleted all the TJC updates to protect beginner users from incorrect firmware flashing. If you are an advanced user and want to test the TJC updates please join the Telegram test group.
The TJC fixes will be available to all users in the next release.

closed 2023-03-22 by mriscoc

mriscoc, 2023-03-22:
TJC Updates, Plot and Mesh Viewer are working
[image]
[image]

CalvinYohannes, 2023-03-23:
does it need a special folder?

ugodnihc, 2023-03-23:
Awesome! Can’t wait for the next release, been waiting to use a custom firmware for some time now :)

gilla013, 2023-03-28:
@mriscoc Is displaying thumbnails for TJC screens parts of this hotfix or related to something else?

mriscoc, 2023-03-28:
> Is displaying thumbnails for TJC screens parts of this hotfix

No, currently TJC screens cannot display preview images.

Ripwords, 2023-03-30:
Will this be included in the next firmware release or do I still need to custom compile a TJC firmware?

mriscoc, 2023-03-31:
> Will this be included in the next firmware release or do I still need to custom compile a TJC firmware?

Currently only by custom compile. I don't know if I will release in the future some precompiled versions for TJC.

gilla013, 2023-03-31:
I did a custom compilation and flashed it and I can confirm these changes also work on my S1 with F4 board.
Any change the thumnail preview images will be fixed? I don't know it if works on stock firmware. I directly flashed this one.

YanivKrit, 2023-04-08:
I have microcontroller "STM32F401RCT6" and when compiling, choosing STM32F401RC_Creality 256K on marling auto build in VSCode, I get the errors:
```
c:/users/aaa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F401RC_creality/firmware-20230408-180820.elf section `.rodata' will not fit in region `FLASH'
c:/users/aaa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 9124 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F401RC_creality\firmware-20230408-180820.elf] Error 1
```

did you get this error too? if so how you fixed it?

mriscoc, 2023-04-08:
> 1. Download the [Source Files](https://github.com/mriscoc/Ender3V2S1/archive/refs/tags/20230312.zip)

You are downloading old sources.

YanivKrit, 2023-04-08:
Thank you, I have downloaded the code from the main git and it works! (with some display glitches but still better)
do you have any suggestion about the UBL?
do you allow me to upload the compiled bin here so people won't have to make all the compilation by themselves?

[image]

mriscoc, 2023-04-08:
> region `FLASH' overflowed by

Lack of program memory space

mriscoc, 2023-04-08:
> do you allow me to upload the compiled bin here so people won't have to make all the compilation by themselves?

Please don't post binaries here, maybe you can do a fork and publish a release page.

drnefarious360, 2023-04-18:
> Thank you, I have downloaded the code from the main git and it works! (with some display glitches but still better) do you have any suggestion about the UBL? do you allow me to upload the compiled bin here so people won't have to make all the compilation by themselves?
> 
> [image]

How did you compile those modifications so that the values ​​come out? and tried to compile your fix from your github and I don't get the values ​​in my tjc screen.

gilla013, 2023-04-19:
Nice work

Dilysir55, 2023-04-23:
> We have the thumbnail preview working for the TJC display!!
> 
> [image]
> 
> It will be available to the all in the next source release.

When will the next release be??

CalvinYohannes, 2023-04-24:
> We have the thumbnail preview working for the TJC display!!
> 
> [image]
> 
> It will be available to the all in the next source release.

when will all this be released?

variableresults, 2023-04-28:
Sign me up for a pre-compiled .bin for the S1 F4 TJC. Mine was a recent purchase, and I suspect there will be more of them out in the wild, so I'm hoping a pre-compiled version will be made since the special compile instructions are a bit beyond me.

mriscoc, 2023-04-28:
For request a pre compiled bin file please check: https://github.com/mriscoc/Special_Configurations#these-special-configurations-and-releases-are-sponsored-by-donors

kobusestas, 2023-04-29:
> I have microcontroller "STM32F401RCT6" and when compiling, choosing STM32F401RC_Creality 256K on marling auto build in VSCode, I get the errors:
> 
> ```
> c:/users/aaa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F401RC_creality/firmware-20230408-180820.elf section `.rodata' will not fit in region `FLASH'
> c:/users/aaa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 9124 bytes
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\STM32F401RC_creality\firmware-20230408-180820.elf] Error 1
> ```
> 
> did you get this error too? if so how you fixed it?
> 
> # EDIT:
> OK so after recreating the configuration file without the UBL there is no change, still no numbers and sometimes the screen returns to be like in the beggining, the steps I have done:
> 
> 1. Download the [Source Files](https://github.com/mriscoc/Ender3V2S1/archive/refs/tags/20230312.zip)
> 2. Download [Special_Configurations](https://github.com/mriscoc/Special_Configurations)
> 3. Open Configurator.pyw and configure Ender3S1->301F4->TJC->BLT & uncheck UBL->T1->no features, then clicking Auto then Set Config Then Generate
>    [image]
> 4. Copy from the generated folder platformio.ini to the source folder
> 5. Copy Configuration.h & Configuration_adv.h to Marlin folder in the source folder
> 6. Install [VSCode](https://code.visualstudio.com/) with [PlatformIO](https://platformio.org/install/ide?install=vscode) and [Auto Build Marlin](https://marlinfw.org/docs/basics/auto_build_marlin.html) extensions
> 7. in VSCode go to PlatformIO in "Activity Bar" and open the source folder
> 8. in VSCode go to Auto Build Marlin in "Activity Bar" -> clicking "Show ABM Panel" -> choosing build for "STM32F401RC_Creality (256K)" because I have STM32F401RCT6 microcontroller and by the datasheet it has 256kb flash size
> 9. copy the created bin file "firmware-date-time.bin" to the printers SD-Card folder "STM32F4_UPDATE"
> 10. power off Printer-> insert the SD-Card to the printer -> power on -> make sure fw was updated (date and time are correct for the version)
> 11. creating mesh 3x3 and tramming - no digits on the circles
> 
> **on step 3 if the "UBL" is checked then on compilation I get the errors above.

I tried this but get the region `FLASH' overflowed by 8948 bytes message with the same config, using the latest code from the repo. Thoughts?

ArkhXI, 2023-05-16:
Recently bought an S1, have the same issue. Other that this everything works correctly, so i guess ill wait for a precompiled firmware.

CalvinYohannes, 2023-05-16:
> Recently bought an S1, have the same issue. Other that this everything works correctly, so i guess ill wait for a precompiled firmware.

yes i also want ready made one

mdemirtas1, 2023-05-21:
i hope the precompiled firmware relased soon for tcj screens

mriscoc, 2023-05-24:
Some special TJC binaries have been released here: https://github.com/mriscoc/Special_Configurations/releases/tag/tjc

mriscoc, 2023-06-07:
> Will the Ender3S1-F1-UBL work for my Ender 3v2?

No, you need a 422 firmware for a 4.2.2 board.

agr8lemon, 2023-06-07:
Ahh ok, So no UBL yet for the TJC for 422 boards yet. Thank you for all your work!

jmealo, 2023-06-08:
Is this just a matter of getting more CI minutes on GitHub?

yurecz, 2023-06-10:
Hello Mriscoc,

Many thanks for all of your work done with the Professional firmware for the Ender!

I have Ender 3 S1 F4 with the TJC display and using the binary Ender3S1-F4-LA-20230524.bin without problems, except that the model preview does not work, and the numbers do not appear in the automatic bed mesh leveling menu as described in this thread.

My question: Would it work to compile the binary by myself for the TJC together with the Linear Advance and BTL without UBL and other features? My concern is that the F4 will not be able to handle the size of the compiled binary. I ask the question because see that the features for F4 always cut in comparison to the F1.

Best regards

mriscoc, 2023-06-11:
> Would it work to compile the binary by myself for the TJC together with the Linear Advance and BTL without UBL and other features?

The different chips used on the various Creality boards have different amounts of flash ROM for firmware storage, this means that different boards will be limited by ROM size to which firmware features can be enabled while still allowing the firmware to fit. The F4 chip has the smallest amount of usable ROM. The maximum firmware size for the F4 is 192k, any compiled firmwares over 192k will not work due to the mentioned size limitations, the precompiled firmwares for the F4 boards have some minor features removed to allow for stuff like LA and UBL and other new marlin features. https://github.com/mriscoc/Ender3V2S1/issues/648