mriscoc/Ender3V2S1 #549: Cannot install Pro FW to Ender S1

ikarus-71, 2022-12-12:
Hello,

I have a Ender S1 with a board STM32F401, firmware v3.0.3.

I correctly formatted the SD card (FAT32, 4096, MBR) to install firmware and copied the related file.

I insert the card and turn on the printer, but it does not read the card and boots up to the stock firmware. Any thoughts?

Thanks

Murat

mriscoc, 2022-12-12:
Please check carefully https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware

closed 2022-12-12 by mriscoc

ikarus-71, 2022-12-14:
Hello, I do not think instructions on the web site is correct, I have to create a folder in SD card as STM32F4_UPDATE and put the firmware there also, and that was the only way it worked for me. This might be an exception for Ender S1 perhaps, but will be good if you can update instructions for everyone.
Thanks 

Murat

mriscoc, 2022-12-15:
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware#manual-installation-for-ender3-s1-with-the-stm32f4-soc

[image]

ikarus-71, 2022-12-15:
Thank you Miguel, had not recognized this.

It is a great firmware, thanks for developing it!

Murat