mriscoc/Ender3V2S1 #552: XY&Z Axis manual setpoint moves

I-wish-2, 2022-12-14:
Has there been any thought to allowing the X Y & Z axis to be manual moved by first dialing in the value you want for the move and then click enter to make the move.  The same as the Extruder does now.  This enhancement would be very helpful when working on the printer and not wanting the axis to just take off when a value is dialed in.  Its hard to stop the mistaken axis move short of powering off the printer.
Regards
Doc'

mriscoc, 2022-12-15:
This was implemented in the latest experimental version and will be in the next public release.

I-wish-2, 2022-12-15:
Greetings Miguel,
Thank you for your reply and making the enhancement changes.  I'm using the Linear advance version and it is working great so far although it requires some very fine tuned input values to get the cornering dialed.  Again Thank you.
Doc'

closed 2022-12-15 by I-wish-2