mriscoc/Ender3V2S1 #556: Sound control in tune menu

medo64, 2022-12-16:
I use "music" to signal start and end of the print. It really comes in handy to know when printer is about to lay the first line so I can verify all is good and even more so to know when bed has sufficiently cooled down so I can get my print off it.

However, sometime I go to bed before print is complete and thus sound will wake me in the middle of the night. Usual workaround is just to turn off sound for that print but I need to remember to do that before I start printing as during printing there is no access to sound control.

It would be awesome if sound on/off checkbox was available in Tune menu similar as to how turning off LCD is. That would allow me to turn off sound when I see that print is going to finish in the middle of night instead of having to do that before the print start.

mriscoc, 2022-12-19:
It may be possible but I'm afraid because of the current size of the Tune menu.

MarkforU, 2022-12-19:
Myself, I would like to see jerk override control to be add in, but I'm sure it's the same problem in the Tune Menu while printing.

medo64, 2022-12-30:
> It may be possible but I'm afraid because of the current size of the Tune menu.

One alternative I could think that wouldn't change size of the Tune menu is to disable sound when `Turn off LCD` is selected. My guess (albeit I might be wrong) is that most people would want LCD and sound off at the same time (e.g. to make it peaceful when going to sleep).

But again, just because that would suit my workflow, doesn't mean that other people would like both turning off LCD and silencing the printer as a single action. :)

medo64, 2023-05-01:
Any thought on `Turn off LCD` option doing both LCD off and muting sound actions?

Assuming that turning off LCD was created in order to remove the disturbance of an LCD screen from environment (e.g. if printer is in living room), having the same option disable sound is not that far fetched. And that would not complicate the issue by adding entries to the Tune menu.

closed 2023-07-12 by github-actions[bot]