mriscoc/Ender3V2S1 #563: Home off center

Rsmyth166, 2022-12-19:
**Describe the bug**
A clear and concise description of what the bug is.
Sprite pro homes off center and probes off the front of the bed

3v2-422-BLTUBL-T13

JEBurroughs, 2022-12-20:
I don't know crap, but here is the process I use to setup mine and the items I check.
Make sure X and Y stops are set so they both apex the front left corner. This is where I set my Park Head to as well and I changed them to 0,0,20. Manually set those first. Then go to  "Advanced/ Physical Settings" and set the Printer Bed size and also make sure your Probe Offsets are correct. You can find those numbers by using the "Prepare/ Move Axis" to find how far you can go. Now Back out to the Main "Advanced/ Mesh Leveling / Mesh Inset" and Set your  Max and Minimum, or it gives you a couple of choices at sets it for you. I am using my Fang Mod with a center mounted BLTouch so my numbers are a bit odd from a normally mounted BLTouch. Now you can probe the Z offset and then Go to the Tramming Wizard

https://www.youtube.com/watch?v=p88xGOVdaYQ

mriscoc, 2022-12-20:
Follow this:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

closed 2022-12-20 by mriscoc