mriscoc/Ender3V2S1 #570: Ender 3 V2 slow screen reaction and then dead after update (?)

Martin-Ox, 2022-12-23:
After updating tot the 20221222 version (4.2.2 board with BLT) the screen reaction after booting was slower then before.
Screen freezed and only thing to do was do a power cycle of the printer. 

Install latest firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3V2-422-BLTUBL-MPC-20221222.bin) on newly formatted micro SD (Fat32)
Reboot the printer
Wait until firmware has flashed.
After correctly flashed firmware, screen is very slow responding and freezes.

**Screenshots**
Just black screen :-( 
**Version (please complete the following information):**
Ender3V2-422-BLTUBL-MPC-20221222

Also no connection through the Serial port. Looks like a broken Motherboard ????

mriscoc, 2022-12-23:
Very weird behavior, I use almost the same configuration without issues. Could be a problem with the uSD card.
Try to re-flash with another memory card.

Martin-Ox, 2022-12-23:
Tried that. But still as black as the night and.

mriscoc, 2022-12-24:
Can you verify that you have a control board 4.2.2 with a STM32F1?

Martin-Ox, 2022-12-24:
Yes it is. See attachement
[image]

yongguiwang, 2022-12-24:
I'm also having this kind of problem and don't know how to fix it

mriscoc, 2022-12-24:
What screen unit have you?

Martin-Ox, 2022-12-24:
DWIN

czy5197, 2022-12-25:
I have the exact same setup and I was having a similar issue. I did the same to flash the latest firmware but ended up just getting a black screen. Now I think the printer is bricked. Trying to flash a previous version of firmware to see if that will fix it.

Martin-Ox, 2022-12-25:
Hi czy5197. 
Please let us know if, **and how**,  you were able to bring the printer 'back to life'.

mriscoc, 2022-12-25:
> DWIN

There are no issues with 4.2.2, STM32 and DWIN firmware. The failure may be related to a problem with the memory card.

czy5197, 2022-12-25:
> Hi czy5197. Please let us know if, **and how**, you were able to bring the printer 'back to life'.

Actually just as I suspected, since the previous version was working before I did the firmware upgrade, I just reflashed the old firmware and it was "back to life." I then proceeded to try flashing the new firmware once again, and I think I might have discovered the issue; but since I stopped testing after it started working, I cannot say for sure. 

Basically the first time I upgraded, I kept all the other print files on the SD card and simply added the firmware file to it. The second time around (also when I reflashed the older firmware), I emptied the SD card, did a reformat (it was always FAT32 though), and only had the firmware file alone on the SD card. I don't recall this being an issue before, but since it worked out in the end, I have no complaint. Hope this can at least unbrick your machine, sadly I'm not sure of the fix for your specific problem.

Martin-Ox, 2022-12-26:
Good for you that you could bring it to life again. I have tried it your way, and with a completely new SD card.
Screen stays black but ... I now have 5 beeps, that could indicate that there is a problem with the thermistor of the hotend. 
Noticed that the thermistor isn't in the top condition as expected. So I have ordered a couple of new ones, that should arive tomorrow evening at the latest. If they arive earlier I will switch the old one with a new one tomorrow, otherwise it will be wednesday. Will keep you informed

yongguiwang, 2022-12-26:
This problem I tried to rewrite the old firmware, successfully wrote and then tried to re-use the latest firmware, but within 30 minutes it was still the boot page, and I found that it was the latest firmware after rebooting

Martin-Ox, 2022-12-28:
Yesterday evening I have receive the v4.2.7 motherboard and a bunsh of thermistors. But, didn't want to rush it. So, this morning first of all a replaced the thermistor. Just in case of.  (even the value messured on the thermistor was around 140k, so that was around the room temperature. )
But..the same issue. Black screen and a few beeps. 
Then replaced the motherboard with the v4.2.7 and flashed the firmware to the 20221002 build for the BLT.
And it now works fine.
So defenetly a firmware issue, related (at least to my) motherboard. 
Could be just a bad luck.

Is there anyone here with a v.4.2.7 motherboard and BL Touch, running on the latest firmware from last week ? And do you have had anny issue ?
Just .... because I don't like to order another motherboard :-)

mriscoc, 2023-01-08:
No issues reported for 4.2.2 boards with STM SoC and tested multiple times.

closed 2023-01-08 by mriscoc