mriscoc/Ender3V2S1 #571: Power Recovery failed to continue print

dfinberg, 2022-12-23:
I had a power outage on my E3 Neo Max. It was a momentary blip, not sure if that made a difference. When I went to check on it, it asked if I wanted to continue the print.  I hit yes, and at that point everything went off the rails.

1) The bed/nozzle preheat was done at the current position, rather than parking first.
2) After it preheated, then it parked and asked me to clean the nozzle and confirm.
3) Then it did that again.
4) Then it continued the print, but still showed "homing" on the screen. I have a video if you want it.
5) After a few extruder lines, it then attempted to drive itself to z=0 which was an absolute disaster since the print was still on the bed. I needed to kill power to rescue it.

I haven't tried to reproduce.  If I have some time I'll try a simpler testcase.

First, it should probably move to parking position before reheating rather than the other order. But obviously there's some more fundamentally broken given the rest.

[Ender3V2-422-BLTUBL-MPC-20221222.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3V2-422-BLTUBL-MPC-20221222.bin)

[IMG_0935](https://user-images.githubusercontent.com/16763150/209360175-343f44eb-41cf-41aa-bda6-6a2bf4c058dc.jpg)

https://user-images.githubusercontent.com/16763150/209360998-6171ddb0-3ae4-4af8-a6cf-9945134f88e2.mp4

mriscoc, 2022-12-23:
Thank you, I will try to do some research on Marlin power outage recovery, but many times it works as intended, it seems that the problems have to do with some large memory cards.

Duplicate: https://github.com/mriscoc/Ender3V2S1/issues/469

dfinberg, 2022-12-23:
Interesting.  I've had the default creality firmware able to continue from this card, which is a 8Gb. Have you seen any difference based on card contents?

BohdanLaba, 2022-12-28:
Exactly the same issue for me:
Ender 3 V2;
4Gb card;
updated firmware to 2.1.3 from 1.0.2,
power outages in my area are pretty common, so with this bug I can`t print anything longer than 8 hours (
No issues with power outage on v 1.0.1 & 1.0.2

mriscoc, 2023-01-05:
How PLR works in this firmware:

https://user-images.githubusercontent.com/2745567/210679567-5708c3bc-fe8c-4f7e-b971-5b2f2cbc8807.mp4

RaffaeleO, 2023-01-06:
Ho avuto diverse interruzione di corrente negli ultimi 10gg con la mia Ender 3 V2 Neo, estrusore creality Sprite Pro e firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin]. Ho continuato la stampa ed è successo un grosso problema.
1. Si è avviato il solo preriscaldamento letto mentre l'Ugello non ha avviato in automatico, ho dovuto reimpostare la temperatura manualmente;
2. Dopo che si è preriscaldato, ha parcheggiato e mi ha chiesto di pulire l'ugello e confermare.
3. Poi lo ha fatto di nuovo.
4. Quindi ha continuato la stampa, ma ha ancora mostrato "homing" sullo schermo.
5. Invece di posizionarsi sul punto dove è stata interrotta la stampa ha tentato di portarsi a z=0, il che è stato un vero disastro poiché la stampa era ancora sul letto, ho quasi rischiato di spaccare estrusore. Ho dovuto spegnere urgentemente la stampante e annullare la stampa al riavvio.
6. Mi è capitato per quasi 10 volte la stessa cosa
Il firmware sarebbe il top se si risolvesse questo Bug e il bug del sensore di filamento.
Grazie

I had several power outages in the last 10 days with my Ender 3 V2 Neo, creality Sprite Pro extruder and firmware [Ender3V2-422-BLTUBL-MPC-20221222.bin]. I continued printing and a big problem happened.
1. Only read preheating started while the Nozzle did not start automatically, I had to reset the temperature manually;
2. After it preheated, it parked and asked me to clean the nozzle and confirm.
3. Then he did it again.
4. Then continued printing, but still showed "homing" on screen.
5. Instead of positioning itself on the point where the print was interrupted it tried to move to z=0, which was a real disaster since the print was still on the bed, I almost risked breaking the extruder. I had to urgently turn off the printer and cancel printing on restart.
6. The same thing happened to me almost 10 times
The firmware would be top if this bug and the filament sensor bug were fixed.
Thank you

mriscoc, 2023-03-13:
The latest version fixed this issue.

closed 2023-03-13 by mriscoc