mriscoc/Ender3V2S1 #574: UBL issues

rlcsoft, 2022-12-24:
Firmware version and release date
- last two versions I have tested 
**Additional context**
Add any other context about the problem here.

mriscoc, 2022-12-24:
Did you maximize and center the mesh insets? https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

rlcsoft, 2022-12-25:
I Followed the instructions, leveled the bed. Reset the firmware and ran G502 and it moved the grid to the center but ubl isn't working on 9x9 with HS off 
that second of my bed 

[image]
[image]
[image]
[image]
[image]

rlcsoft, 2022-12-25:
[image]
[image]
[image]

rlcsoft, 2022-12-25:
[image]
[image]

rlcsoft, 2022-12-25:
Can you reach out to me via email please : rlcsoft@gmail.com,

mriscoc, 2022-12-25:
I just saw an issue introduced by recent changes in the G30 code of Marlin that could have an effect in the determination of the probe positions. I will need some time to investigate if it is related.

mriscoc, 2022-12-26:
I did some probe positioning tests and were successful, mesh is correctly centered after the maximize/center procedure.

rlcsoft, 2022-12-26:
I swapped CR Touch for BL Touch, testing now with UBL special config compile. I noticed different readings.

Will keep you updated.

rlcsoft, 2022-12-26:
Results:
The Swapped to a BL-Touch fixed the readings on the bed(tried another new CR-Touch) 
9x9 mesh area was small, 7x7 was more spread out went further on the bed. 

Need advise on spreading the probe zone to 15mm from edges.  How can 9x9 be fixed. 

[image]
[image]

mriscoc, 2022-12-26:
Some probing positions are unreachable and that depends on the probe offset. With lower probe offsets you must be able to decrease the margins and increase mesh area. If you don't need a centered mesh area just select maximize area, the printer will use the maximum reachable probing area available.

rlcsoft, 2022-12-26:
Cr touch was confirmed issue.

daninden, 2023-01-03:
> Cr touch was confirmed issue.

is it easy to swap the cr touch for a bl touch?

Tempestine, 2023-01-22:
I've been having issues with UBL on my Ender 3 V2 with a 4.2.7 board since updating to the lastest professional firmware (https://github.com/mriscoc/Ender3V2S1/releases/download/20221222/Ender3V2-427-BLTUBL-MPC-20221222.bin) firmware. The mesh would map strange and when I printed a bed level test, one side was squished and the other side would not stick. I thought it was perhaps not applying the mesh. I think I solved it by compiling my own firmware from the source. The mesh looks totally different now. I think there may be something wrong with some of the precompiled firmwares.

mriscoc, 2023-01-25:
For Q&A please post in [Dis](https://github.com/mriscoc/Ender3V2S1/discussions)

closed 2023-01-25 by mriscoc